TinkerGnome/Ultimaker2Marlin #2: LEDs flickering

Alex9779, 2015-04-05:
Hi,
I recently tried out your special firmware v15.03 because I wanted to have that extended status information when printing.
I did not change anything in the setting after installing it other than switching the display to geek mode.
After two successful prints I realised that during the entire process and in idle the LEDs are flickering. It is not much but it is there. So I switched back to the original firmware because I also read about some problems in the past with really heavy flickering of the LEDs to see if it was there but I never realised it.
But with original firmware there is no flickering.
Any idea why the LEDs do this?

Regards,
Alexander

TinkerGnome, 2015-04-06:
Hi Alexander,

at least it's not an common issue. I never heard something like this before...
First you could check the settings of the sleep timer.

Another sneaking suspicion: in contrast to the stock version the PID-controlling of the printbed temperature is activated - that means: slightly higher electromagnetical radiance. Perhaps this has an influence to other PWM-controlled components?
Would you like to test a version with the standard "bang-bang" mode for the printbed temperature?

Greetings from the TinkerCave.

Alex9779, 2015-04-07:
Hey TG,
I did a little test and you are probably right with your second guess.
I reinstalled 15.03, power cycled the printer and the LEDs are steady. Then I checked the sleep timer settings, all I changed there was the time on both to 60 minutes and to 30% but the flickering is not when sleeping.
So I heated up the bed to 60°. During heat up no flickering but when the bed reaches the temperature and temp controlling starts the LEDs flicker. TO clarify the flickering is not like on off or huge brightness differences, it is more like about 5% brightness and a very high frequency, at least 20 Hz.
When I set the temp to a lower value no flickering until the temp is reached, same when increasing temp.
So the flickering is connected to the PID-controlling somehow. Is that so important or so much better than the standard way, can you hook me up very short why you want that PID?

Regards,
Alexander

TinkerGnome, 2015-04-07:
There were some indications that PID controlling of the heated bed can improve the surface quaility.
Sources are (for example) [here](http://umforum.ultimaker.com/index.php?/topic/6907-horizontal-banding-on-um2/page-2#entry75660) and [here](http://umforum.ultimaker.com/index.php?/topic/6613-firmware-surface-kwality-bug-1407-vs-1403/page-2#entry75597).
To be honest - i never made such excessive comparisons for myself...

TinkerGnome, 2015-04-07:
I've just uploaded a version with disabled PID for the bed temperature control to the release page:
[TinkerGnome-15.03-printbed-bang-bang.zip](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/download/V15.03/TinkerGnome-15.03-printbed-bang-bang.zip)

The source code is located in a new branch:
[printbed_pid_test](https://github.com/TinkerGnome/Ultimaker2Marlin/tree/printbed_pid_test)

Alex9779, 2015-04-08:
Well with that version, what a surprise ;), the flickering is gone...

Alex9779, 2015-04-15:
For me this is fixed with the new branch.
Will you keep that new one up an updated?
I didn't have any issues so far with the bang-bang bed and the comparison of two test prints, one with bang-bang the other with PID and flickering LEDs didn't show any differences that I noticed...

TinkerGnome, 2015-04-19:
preliminary - i will keep both branches up to date, but i think  - on the long term - it would be better to make it a switchable option.

closed 2015-04-19 by TinkerGnome