TinkerGnome/Ultimaker2Marlin #4: OctoPrint Through USB Nozzle Shutting Off

scottdanesi, 2015-10-26:
Hello,
I am using the latest version of the TinkerGnome software v15.10.1 on my Ultimaker 2.  When I print over USB using OctoPrint, the heater will shut off once the print starts.  If I manually (through OctoPrint) turn it back on, it will shut off again on the next line of GCode. 

This is pretty strange as the heated bed stays properly at 60deg the whole time like it should.

Below is the startup script that I am using in my GCode along with the first few lines of my print.

G90
M82
M106 S0
M140 S60
M190 S60
M104 S218 T0
M109 S218 T0
;Modified By Scott Danesi
M907 E1400 ; increase extruder current
G28 ; home all axes
G1 X20 Y10 F3000 ; bring extruder to front
G92 E0 ; zero the extruded length
G1 Z10 ; lower
G1 E19 F200 ; purge nozzle quickly
G1 E26 F60 ; purge nozzle slowly
G92 E0 ; zero the extruded length again
G1 E-5.5 F400 ; retract
G1 X190 Z0 F9000 ; pull away filament
G1 X210 F9000 ; wipe
G1 Y20 F9000 ; wipe
G1 E0 ; feed filament back
G92 E0
G1 E-4.0000 F1500
G1 Z0.200 F1002
; layer 1, Z = 0.2
T0
; tool H0.200 W0.624
; skirt
G1 X157.081 Y5.622 F6000
G1 E0.0000 F450
G92 E0
G1 X157.338 Y5.494 E0.0056 F900
G1 X160.929 Y5.243 E0.0760
G1 X161.201 Y5.334 E0.0816
G1 X181.942 Y23.364 E0.6193
G1 X182.067 Y23.598 E0.6245
G1 X182.488 Y26.883 E0.6893

Please let me know if there is anything else you need from me.  I am sure there will be.  :)

--Scott

TinkerGnome, 2015-10-26:
Hi Scott,

thanks for reporting.
I have a first idea... - possibly the heater timeout "kicks in" during usb printing. Could you please try to disable the timeout (as a temporary workaround):
Advanced -> Preferences -> Heater timeout (decrease the value until it shows "off")

scottdanesi, 2015-10-26:
Looks like that is the issue.  I am now able to print using USB with the Tinker Firmware.  If there is anything else you would like me to try, please let me know.  I am glad to help.

Thank you,
--Scott Danesi

TinkerGnome, 2015-11-02:
Scott, this should be fixed with [release 15.10.2](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V15.10.2).
Can you try it with this version? (...but re-enable the heater timeout first... :smiley:)

scottdanesi, 2015-11-02:
Absolutely.

 

Stay tuned for results.

 

Thanks,

 

Scott Danesi

scott@scottdanesi.com

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2016.0.7163 / Virus Database: 4455/10933 - Release Date: 11/02/15

scottdanesi, 2015-11-02:
I just checked the latest version printing over USB and it worked great.  Thank you for getting this fix in place so quickly.

TinkerGnome, 2015-11-03:
Goody... :smile: - thanks for testing - have fun!

closed 2015-11-03 by TinkerGnome