TinkerGnome/Ultimaker2Marlin #6: Thanks for M42

rooiejoris, 2015-10-29:
Hi,

First of all thanks for the M42 fix, my students now where able to combine two machine again like 2 years ago. Thanks!!
Since you do quite some things with the firmware, I was wondering how difficut it is to add babystepping to the UM2 firmware.... For the Z-Unlimited [www.rooiejoris.nl] I made custom firmware for the UM original and the UM2, but for the UM original I was able to add the "babystepping" feature. This is a sort of live offsetting during printing. Why would you need that... : ) Since you can print very large objects, sometimes you get failures. Then you maybe don't want to restart a 50+ hour print. Since the Z-Unlimited I master resuming prints pretty well, and the small tweaks in X and Y are then very usefull.
I implemented it in Marlin for the original, but I am a product designer not a programmer... : )
The initial code was made by bernhard kubicek, initially for the Kamermaker for the 3D print canal house. One part was a M192 gcode command, but we figured that it takes to long if the offset was planned in the buffer. The babystep is a real change in the stepper... [as far as I understood...]

cheers \ joris

www.rooiejoris.nl
www.facebook.com/europerminutedesign

TinkerGnome, 2015-10-30:
Joris,
i would never want to restrain your creativity :smiley: - i would like to assist.

So actually - "babystepping" is an "x/y/z shift" with small amounts during printing, right?
Basically you want to "lie" a bit about the current position of the appropriate stepper motor to correct small misalignments.

I have not seen such a function "life" (yet) . But (aside from the menu changes) the implementation would be quite similar to the one of the UMO. Do you know any sources for this?
...and just to be sure... do you would like to see this ability here in this fork, or do you need assistance with your own somewhere...?

rooiejoris, 2015-10-31:
This functionality is present in the UMO formware, and i think that it is 'just' a copy paste from stepper.cpp, stepper.h temperature.cpp and temperature.h for the core
and something from ultralcd.cpp for the menu structure
[and for the completeness something in configuration.h and or configuration_adv.h]

This is a pretty high level user functionality, but if you get good in resuming prints, it is very handy to have. You don't ave to start a >20 hour print again, you 'just' resume...

about the repository, don't know what is smart, you may edit directly direct in github/rooiejoris/ultimaker2_zunlimited if you want [let me know what I should do then, I am not so familiar with git and github]

cheers / joris

TinkerGnome, 2015-10-31:
I found out right now - babystepping is part of recent version of Marlin. Sorry  for my ignorance - i withdraw my question.... :smiley: 
As far as i can see you've already copied some parts from there. The crucial calls of the "babystep" function from inside the TIMER0 interrupt are missing, and you would need some addtional entries for the tune menu. This should be feasible, i will take a look at it.
If i get it, i will open a pull request for your fork and you can merge it - if you like (this should be the best way).

TinkerGnome, 2015-11-03:
Joris, i was not able to fork your fork, because all of these forks have the same name (intentionally).
But i made a copy and the needed changes for a working babystepping.
You can find it here: [https://github.com/TinkerGnome/ultimaker2_zunlimited.git](https://github.com/TinkerGnome/ultimaker2_zunlimited.git).

Some remarks:
Your master branch didn't compile atm, i assume that you want to remove (resp. replace) all of the stuff for the "first run wizard" and "change material wizard"? I had to remove most of it and was able to compile. Babystepping works great... :smiley: I tried to automatically "tune" the sensitivity according to the axis steps per unit. You can modify it in [UltiLCD2_menu_print.cpp](https://github.com/TinkerGnome/ultimaker2_zunlimited/commit/62a4f5b9e5f83b759a8d8bf5b533b7ae486e7ee3#diff-41dee0ad5d891d59fab5c856765b9c1eR629) for your needs.

rooiejoris, 2015-11-03:
THANKS, now testing...
"Edit: i also removed the remains of the "heater error" checks. I'm not sure if this was your intention, just keep it in mind."
don't know what that is, i was already happy that I could change the firmware for rotating the LCD, flipping the X axis, prevent homing after the print is finished, keep power on the Z-motor, change steps per unit, make new startup wizard [which didn't work completely] et cetera...

cheers / joris
btw have we met before...?! I would like to send a personal email...

TinkerGnome, 2015-11-03:
Well, you have removed these lines from the Configuration.h

    // Check if the heater heats up MAX_HEATING_TEMPERATURE_INCREASE within
    // MAX_HEATING_CHECK_MILLIS while the PID was at the maximum.
    // If not, raise an error because most likely the heater is not heating up the temperature sensor.
    // Indicating an issue in the system.
    #define MAX_HEATING_TEMPERATURE_INCREASE 10
    #define MAX_HEATING_CHECK_MILLIS (30 * 1000)

These are needed for the heater check - introduced by David (Daid) in 12/2014 (i think). I just wondered, if you had any problems with it and removed this intentionally, or not...?

> btw have we met before...?! I would like to send a personal email...

That's unlikely - i rarely go out of the tinkercave... :yum:
I'll send you a contact request via http://www.rooiejoris.nl/contact.

rooiejoris, 2015-11-04:
>> These are needed for the heater check - introduced by David (Daid) in 12/2014 (i think). I just wondered, if you had any problems with it and removed this intentionally, or not...?

well... I didn't remove it intentionally...
There are only a few things I did intentionally, but when I broke functionality I am pretty sure that it was not on purpose....

>> i assume that you want to remove (resp. replace) all of the stuff for the "first run wizard" and "change material wizard"? I had to remove most of it and was able to compile. 

I compiled and uploaded the firmware and the first run wizard for Z-Unlimited was still in. I used the wizard to change the z-step [at least i tried] tot the right amount of steps and store it in memory.
The material change wizard should be in there, i didn't change that on purpose.

cheers / joris

TinkerGnome, 2015-11-07:
Joris, i tried another approach:

I took the [current master branch from Ultimaker](https://github.com/Ultimaker/Ultimaker2Marlin) added the babystepping and merged in your changes for z-unlimited [from here.](https://github.com/rooiejoris/ultimaker2_zunlimited)

I checked the menus with the simulator but (obviously) i cannot test the z-probing and the axis movements. So be careful... :smile:

If you have another working version, just compare (and merge) the changes from stepper.cpp, temperature.cpp and UltiLCD2_menu_print.cpp

Have fun!

rooiejoris, 2015-11-07:
ok, not completely clear for me at the moment, but I think you mean that I can use the zunlimited_joris branch or not?
if so I will try to keep my version up to date with that branche.

cheers / joris
[btw I don't use z-probing for the z-unlimited or do you mean the homing direction...?!]

TinkerGnome, 2015-11-08:
> but I think you mean that I can use the zunlimited_joris branch
in short - yes, that's what i mean...

longer explanation:
Your fork doesn't compile (at least for me) - so i assumed that this are not the same sources that you use for your "production code".
You added some [stuff like this.](https://github.com/rooiejoris/ultimaker2_zunlimited/commit/86a0c091e65ad8a3ff9c179f59139ac2a2ff0743#diff-8270513bcf2a548cd418cdfe123f27d3R344)
...and on the other hand some important #defines are missing...
This was a bit confusing to me and you also said, that it was not (all) on purpose...

That's why i decided to start from scratch with the latest master branch from Ultimaker. I added your changes to switch the direction of the axis and the display - and also the additions for "babystepping" (copied from the MarlinDev repository).
And here it is - you can give it a try as it is, or merge it with your production code (or whatever... see above).

Do you have any plans to adopt the tinker firmware for z-unlimited? This would be a lot easier for me... :smile: (just email me if required)
Good luck!

closed 2015-12-12 by TinkerGnome