TinkerGnome/Ultimaker2Marlin #7: Heater Error Timeout

gregerspoulsen, 2015-10-30:
Hi

I just installed an E3D hotend on my UM2 and is now using your firmware to adjust the build area. First of all I would like to thank you for a really great firmware it's a huge improvement!

I do however have one slight issue. Due to the larger thermal mass of the hotend and a larger distance between the sensor and heater, there is a larger delay between the the time when the heater switches on and the temperature reading increases. In particular if the head is cooling down when the heater is switched on, this typically results in a heater error which is a bit annoying. I could of course just compile the firmware myself with a larger timeout, but as I imagine this could affect other users too I was thinking it might make sense to add a menu item to set this timeout value? Just a thought, and once again thumbs up for a great firmware!

TinkerGnome, 2015-10-30:
Thanks, but you're much too kind.... I've not reinvented Marlin after all - i only made some minor changes to the user interface...
Regarding the heater check i agree with you (somewhat...) - one can never have to much preferences... :smiley: Let's put it on the todo list.

TinkerGnome, 2015-12-12:
The allowed period of max. heater power is adjustable now [V15.12](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V15.12) - (Advanced -> Preferences -> Heater timeout)
Would you like to make a test with this version?

gregerspoulsen, 2015-12-15:
I actually installed 15.11.RC1 some days ago to test it out, and it works perfectly, I should probably have mentioned. Thanks a lot - if I'm to kind so are you :smiley:

closed 2015-12-23 by TinkerGnome