TinkerGnome/Ultimaker2Marlin #8: Heater control not OK

EldRick, 2015-11-05:
I've got an issue with heater control. If I set Cura to send RepRap gcode to my UM2 with your 15.10.2 and earlier releases, fan control works fine. M106 commands are sent, and work correctly.
However, if I set Cura to use UltiGcode, fan control screws up: if I set 20% fan to come on at layer 3, I get 1% at layer 2, and 2% at layer 3.
The only oddity I can see is that recent Cura versions send M106 codes with non-integer S-values, i.e. 
M106 S40.39000
However, changing the values to integers does not seem to work.

TinkerGnome, 2015-11-05:
EldRick,
...either you missed my query on the ultimaker forum, or i missed your feedback... :question: 
...in short:
For UltiGCode the value of the M106 command will be multiplied with the fan percent of the current material setting (at the printer)... A value lower than 100% there would explain your symptoms. Have you checked the material settings on the printer?

EldRick, 2015-11-06:
I’ll be damned - I never noticed that behavior in a year of owning the UM2. 

Yes, I had the fan set at 20% in the UM2 settings, and I have verified that 100% and 0% work as you state with UltiGcode. I’ve never seen that documented anywhere.

Thanks!
Rick

TinkerGnome, 2015-11-06:
You're welcome! Happy printing!

closed 2015-11-06 by TinkerGnome