TinkerGnome/Ultimaker2Marlin #9: Stepper motors power after filament change

Alex9779, 2015-11-21:
Hi,
I think this is also an issue of the stock firmware but since I only use yours now I post it here...
When I change filaments using the change option I often wanna do the cold pull so I use "Change" to remove the filament then I cancel and do my stuff.
After that I use the "Insert Material".
I realised that after the change the stepper motors are not powered off like they are when a normal print is finished or cancelled. I always have to go deep down the advanced menu and use the option to power off the steppers to be able to move the print head.
Foul did be possible to implement the power off after material change too?

Kind regards,
Alexander

TinkerGnome, 2015-11-22:
Hi Alex,
i made some changes on this for the latest [test release](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V15.11.RC1) - would you like to try it?
...or as a workaround: you can also use the "Move material" function to remove the material. This should disable the stepper motors afterwards.

Alex9779, 2015-11-24:
Hey TK,
now it works. Didn't figure out what you changed in that monster commit which did that but I works :)

closed 2015-11-24 by Alex9779

TinkerGnome, 2015-11-24:
Great - thank you for the feedback!

I think it was [here](https://github.com/TinkerGnome/Ultimaker2Marlin/commit/b032c741ecbc1c0b49ac767b601d340946b7859b#diff-1ff1af4d39e9fc1759e645f866ea3545R273) and [here](https://github.com/TinkerGnome/Ultimaker2Marlin/commit/b032c741ecbc1c0b49ac767b601d340946b7859b#diff-56df5a3701627b345f418a43ff798b74R40) (additional M84 commands)