TinkerGnome/Ultimaker2Marlin #10: Pause icon

pfer, 2015-11-23:
Hello TinkerGnome,
awesome work, i can't go back to the original firmware!
Just a suggestion maybe you should replace the pause icon by a toothed wheel or a tool (I know it's tiny): when you click pause you doesn't pause directly, you go the settings menu where you can pause or abort or tune. 
Do you think it's possible to setup the led brightness directly from your main print screen (maybe on a page 3). It's a feature a regular use.
Keep up the good work!

TinkerGnome, 2015-11-24:
Thanks for the suggestion,
finding a good looking icon in 8x8 pixels is always a challenge... i you have an idea...don't hesitate :smile: 
Perhaps it's the best to tentatively use the "menu" icon for this (the same as "Tune")?

The led setting is noted - there is free space on the bottom line... let's see...
Do you really regulate the brightness frequently, or would it be better to have a simple on/off switch?
And as a supplementary question: Did you noticed the "Sleep timer" options? The leds can be switched off after a defined period of inactivity.

pfer, 2015-11-25:
Hello TinkerGnome,
for the icon you can do a tool ;) 
[image]

For me a simple on/off switch will do the job, but you can also do a 0% to 100%.
[image]

I notice the sleep timer, it's a great feature! But most of the time I switch off the leds and turn it on when I want to see if the print is fine or not.

Blizzke, 2015-11-25:
The sleep timer is great, but whatever time I set it to, I usually arrive 5 seconds after that to check the print and turn the leds on again :) So a "sleep mode" quick link (not just leds but display and leds) would be quite awesome for me as well :+1:

pfer, 2015-12-12:
@TinkerGnome Thanks for the update! it works nicely.

closed 2015-12-14 by TinkerGnome