TinkerGnome/Ultimaker2Marlin #11: SD Card Folder Navigation AFTER print

Alex9779, 2015-11-24:
Hey TK,
I got another one.
I am using sometimes a folder structure on my SD cards. Now I realised that AFTER a print when I go back to the print menu I see the content of the root folder but the first item is not tabled "RETURN" but "BACK". So when I hit that I am one folder up from the file is previously printed.
It gets more strange when I again navigate to a sub folder and the go up again. Then things really screw up.
I looked in the code and I think it has something to do with caching the current folder and jumping back to the folder and selection the previous printed file. Something is not cleared correct after a print happened.
But unfortunately I am not so into C++ and Marlin yet that I can figure out myself an create a pull request...
Can you have a look?

Kind Regards,
Alexander

TinkerGnome, 2015-11-24:
I assume that you're on the right track. The SD card support in preceding Marlin versions is quite fragile...
I put it on the list...

Alex9779, 2015-12-08:
For me it would be enough if after a print the menu start at the root again.
As I understand the code it should mark the last printed file but this even does not work if a file in root is printed.
So just removing all the jump to the last printed file stuff would be the best IMHO...

TinkerGnome, 2015-12-12:
I agree with you, it always opens the root folder, but does not reset the "workDirDepth". This should be fixed now [V15.12](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V15.12)

Alex9779, 2015-12-12:
Cool. Thanks. Will check later and report. Will also have a look at what you changed. Hopefully it will help me understand the code better...

Alex9779, 2015-12-12:
Works :)

closed 2015-12-12 by Alex9779