TinkerGnome/Ultimaker2Marlin #12: Preferences -> Temperature Control menu buggy

Alex9779, 2015-12-08:
Hi TG,
I recently went through the preferences and tried to access the Temperature control menu. When I click the menu entry nothing happens. It seems I am in the menu but do not see any changes. When I click the button again I am back in the preferences menu and can move the cursor again there. When I make a scroll with the wheel for about one or two ticks and click then again I seen the settings in the menu. Then the back button do not bring me back to the preferences menu, I see still the temperature control menu.
Seems there is some bug in the menu bathing there...

Regards,
Alexander

TinkerGnome, 2015-12-08:
ooops, thanks for testing the single extruder version... :smile: 
I'll take a look at it.

TinkerGnome, 2015-12-12:
This should be fixed with [V15.12](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V15.12)

Alex9779, 2015-12-12:
Will check that too... You are the best...

Alex9779, 2015-12-12:
Works :)...

closed 2015-12-12 by Alex9779