TinkerGnome/Ultimaker2Marlin #13: Printing Time

ericspakauskie, 2016-01-11:
Hi there TinkerGnome,

I don't know if it's just me or I don't know, but I have noticed the difference between Cura printing time and the Ultimaker printing time. Chura shows 19hours 40 minutes, Ultimaker shows 35hours. I am currently using the latest firmware 16.01 for the ultimaker 2.

Thanks for this amazing firmware!

-Eric

Blizzke, 2016-01-11:
Cura is always the correct one. It knows the complete print path and can determine pretty much up to a second how long it will take. On the other end: determining printing time on the UM2 has always been vastly inaccurate. If I'm not mistaken it simply uses the current file pointer and how long it took to get there. Based on that it will project how long it will take to finalise the entire file. Which is why it changes so much if you have a lot of .gcode the results in very small movements or for example a line that is drawn at an extremely low speed.

There have been suggestions already to include ETA data into the gcode as comments so that the printer can show accurate timings, but so far no one has done this as it requires changes both in cura and in the firmware.

ericspakauskie, 2016-01-11:
Thanks Blizzke for the info. I just noticed it since I updated on the latest firmware.

TinkerGnome, 2016-01-11:
It's even worse, if you don't use UltiGCode... :smiley: 
I'm afraid that a better prediction would overstrain the poor Arduino. The displayed percentage is also only the file position.

closed 2016-01-11 by TinkerGnome