TinkerGnome/Ultimaker2Marlin #14: Tinker_16.01 LCD freeze

darkdvd, 2016-01-21:
During print, the LCD panel freeze.
Printer : UM2
Tinker : Tinker_16.01

TinkerGnome, 2016-01-23:
Hi @darkdvd,
that's a very "economical" report - no further informations:question: 

- printing from sd card or USB?
- and the print stopped or just finished normally?
- is it related to the "Screen timeout" settings or something else?
- is it reproducible - and if so: what do i have to do to reproduce it?

darkdvd, 2016-01-23:
. printing from card
. print finished normaly
. not related to "Screen timeout" setting
. not reproductible : I do 6 prints from the same object and no problems

note : during the "LCD freeze", the button work. If I put the button, the color of the button change. I put the button and the color change but just after, no color.

darkdvd, 2016-03-02:
Hi !
Strange situation.
Got a new UM2 and the first thing I do it's to use tinker firmware 16.01 (of course !) but this morning, I have a problem with the "LCD Freeze" too on this printer... Strange.
The file I print was different from the precedent problem with the other printer : a single object for a 30 minutes print...button work but the LCD was stopping 4 minutes before the end of the printing.
David

TinkerGnome, 2016-03-03:
ahhh - it's epidemic - stay on distance! ...  ;-)
I remember only one small change in 16.01 that is (perhaps) related to lcd_update. I suggest: let's revert this for the next version and see what happens...

darkdvd, 2016-03-03:
I will do that but for information, after a freeze, the LCD is comming back during the print !
Thank you TinkerGnome !