TinkerGnome/Ultimaker2Marlin #15: Parameterize to fixed values ...

gpb01, 2016-01-24:
Hi Tinkergnome,
I report here what I already asked on the UM forum ... :-)

... Is it possible to add two parameters (maybe) in Advanced -> Preferences -> Motion Settings -> *unload/load speed* ...

The two parameters should allow to set, by the Menu, the two values currently fixed by these two constants in the firmware (file ULTI_LCD2_MENU_MATERIAL_H):

#define FILAMENT_REVERSAL_SPEED
...
#define FILAMENT_INSERT_FAST_SPEED

The reason for this request is to make it easier to use third party extruders that, often, requires different values (*e.g. BondTech QR Extruder*) without the need to re-compile the firmware. 

Thanks in advance,

Guglielmo

TinkerGnome, 2016-03-20:
With the [latest release](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V16.03) the filament change speed  depends on the steps/mm of the feeder. This should work with a wide range of feeders. If you like, test it out.

gpb01, 2016-03-23:
Hi TinkerGnome,
thanks a lot, as soon as possible I will test ;-)

closed 2016-07-31 by TinkerGnome