TinkerGnome/Ultimaker2Marlin #17: babystepping question [not really an issue...]

rooiejoris, 2016-01-26:
Hi,
Just tried the firmware on a UM2 [not the z-unlimited version], really really nice...!!!!!
I actually wanted to use the baby stepping function for a project where I have to resume some prints. But where can I find the function in the menu...!?

BTW I also saw the resume print option, that is even better then I had in mind for this project, but still want the possibility to make small adjustments. tip for others when using the resume: remove roughly the previous layers in the gcode, that saves quite some seconds...

cheers / joris

TinkerGnome, 2016-01-27:
I have to admit, it's a bit hidden...
On the printing screen: move the cursor to the Z-position to access the babystep menu.

rooiejoris, 2016-01-28:
hmmm... I can't move to the Z-Position when printing...?!
And how can I acces X and Y...?!

cheers / joris

TinkerGnome, 2016-01-29:
Yep, sorry i didn't had this in mind...
I use the Makefile all the time and set those #defines on the command line
(and do not think about it anymore...)

rooiejoris, 2016-01-29:
still the other question, how to babystep the X and Y...?!
btw is it an idea to place a hex file in the repository as well, I think I
can convince way more people to use this firmware then. I notice that
compiling is quite a threshold for many designers...

cheers / joris

rooiejoris, 2016-01-29:
>> still the other question, how to babystep the X and Y...?!

please forget this stupid question... : ) mixing up different things

cheers / joris

TinkerGnome, 2016-01-30:
> btw is it an idea to place a hex file in the repository as well, I think I can convince way more people to > use this firmware then. I notice that compiling is quite a threshold for many designers...

Well, you can commit binary files to the repository, but the download from GitHub is a bit unconvenient (the keyword is: "raw" mode).
I use the [Release section](https://github.com/TinkerGnome/Ultimaker2Marlin/releases) to provide the compiled hex files. For a release you can upload as many additional files as you want.
(I'm not sure about the allowed file types, but *.zip is always possible)

rooiejoris, 2016-01-30:
Ah. That's even better!
I will point in that direction then.

Cheers / Joris

closed 2016-07-31 by TinkerGnome