TinkerGnome/Ultimaker2Marlin #20: Temperature overshot when starting cold extruder

Fraeser, 2016-02-18:
There is an variable called "PID_FUNCTIONAL_RANGE "

IT is set to 1000, shouldn't be higher then 20.

If it is set too high you have massive temperature overshot when heating cold or inactive extruder.
Because of the big delay from starting the heater to its final temperature you get in PID calculation an huge I-value.
Thats responsible for temp overshot.

TinkerGnome, 2016-02-26:
I'm pretty sure, that i never touched this value - but it's a good point. I'll take a look at it, thanks!

TinkerGnome, 2016-03-03:
Hi Fraeser,
i made some experiments - a value of 15 seems to be a good range for the Ultimaker hotend. The overshot is reduced to 1-2 degrees. Huge improvement!
You should repeat your report at the [Ultimaker master repository](https://github.com/Ultimaker/Ultimaker2Marlin)

closed 2016-03-16 by Fraeser