TinkerGnome/Ultimaker2Marlin #21: ultimaker2go heated bed suppport

ARKopp, 2016-02-25:
is there already the feature that you have a heated bed in the firmware?

TinkerGnome, 2016-02-26:
AFAIK, George made a version for the 3DSolex heated bed, you should contact him for details.
Here is the link: http://gr5.org/hb/

ARKopp, 2016-02-26:
ok thanks I already got that one. I thought yours was different.

TinkerGnome, 2016-03-20:
I added a variant for the UM2go with heated bed. I don't own this model, would you give it a try and report any issues here?
Here is the [release](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V16.03).

ARKopp, 2016-03-27:
works great! just checket it.

ARKopp, 2016-03-27:
no issues so far with it.

ARKopp, 2016-03-27:
now I got an issue. When I changed the material then you get out of the menu. then the print is still printing and I am in the inital menu.

ARKopp, 2016-03-27:
also when changing the material it starts again and then pushes material for priming out right on top not on the side before starting again.

ARKopp, 2016-03-27:
ok after changing the material the menue switches in the menue from the beginning to stop the print you have to turn off the printer.

TinkerGnome, 2016-03-27:
Yep, you're right - the pause menu is broken with this version. This applies to all variants...
I'll fix it in the next days - if i find some time...

ARKopp, 2016-04-07:
Hi,

do you have the firmware on github as well so I could do some adjustments
there?

thanks

Andreas

TinkerGnome, 2016-04-08:
It's the normal branch [geek_UM2go](https://github.com/TinkerGnome/Ultimaker2Marlin/tree/geek_UM2go), compiled with these defines:
`TEMP_SENSOR_1=0`
`TEMP_SENSOR_BED=20`
`EXTRUDERS=1`
`FILAMENT_SENSOR_PIN=30`
`BABYSTEPPING`
`HEATER_0_MAXTEMP=275`
`HEATER_1_MAXTEMP=275`
`HEATER_2_MAXTEMP=275`

ARKopp, 2016-04-08:
ok cool thanks. somethime in the future I will do it myself.

Did you have time to fix the stop error?

TinkerGnome, 2016-04-09:
The menu order (for the pause menu) should be fixed now. Regarding the the material change (mid print) - i hope it works as before, but i had not enough time to make tests.

closed 2016-07-31 by TinkerGnome