TinkerGnome/Ultimaker2Marlin #22: Tinker_16.1 printing issue (maybe incorrect read from sd)

vaques, 2016-03-02:
Dear,

My UM2 clone with Tinker 16.1 has some issues during printing like:
- Skipping layers
- The head start moving in right or left direction with no sense
- Error: Tried printing out of printing area

I did some test (with no success):
- another SD;
- Changed ribbon cables
- Changed motherboard :)

Downgrading to Tinker_15.11.RC1, the realease before new SD library, the UM2 works well.
 
In the attachment the stl and gcode that doesn't work (sliced with cura 15.04.4, it's a part of openrc project...)

[Example.zip](https://github.com/TinkerGnome/Ultimaker2Marlin/files/155816/Example.zip)

Fraeser, 2016-03-04:
do a factory reset in machine prefrences, and the machine is working.
Don't use a SDHC card must have a SD card

Greetings Fraeser

vaques, 2016-03-04:
I did a factory reset but I'm not sure to use SD instead of SDHC. Surely yesterday night I printed the same piece in the example without issue (2 hours of printing) with Tinker_15.11.RC1. I'll try this night with Tinker_16.1 and a SD card.

Regards,
Alberto

Fraeser, 2016-03-05:
I had also trouble one day with "printing out of printing area".
Looked in Ultimaker forum for information about SD card. 
They say use SD, no SDHC card

https://ultimaker.com/en/community/4193-sdhc-card-not-supported-in-ultimaker-2

TinkerGnome, 2017-01-01:
@Fraeser, @vaques 
V16.12 uses the originally SD library - can you report, if the issue persists?