TinkerGnome/Ultimaker2Marlin #23: Ultimaker 2+ Firmware

scottdanesi, 2016-03-12:
Hey Mark,
I have been using your firmware for a while and LOVE it.  I just got an Ultimaker 2+ and would love to use your firmware on it as well.  Any plans to make a version for the UM2+?

Thanks man!
--Scott

korneelb, 2016-03-14:
yes please..

so I found a serious annoyance.. 
with the original PT100, there was a serious issue with measurements (hence the improvement..) but if you want to have some fun, attach a Fluke Thermometer and you would find out that when printing at 255 on an UM2, you were actually printing with around 270 to 275 in the head.. 

now I've upgraded my UM2s to UM2+ printers.. and I can't go beyond 260 real degrees anymore.. I need to go to 275 or maybe even more.. 

so yes.. please..

TinkerGnome, 2016-03-20:
[Here you go...](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V16.03)
max. temperature is 300°C.
Keep in mind that i only have one single printer, i cannot test all variants by myself...
Just report any issues here.

scottdanesi, 2016-03-20:
I can test the Ultimaker 2+ if you need.  Just let me know.  The extruder stepper ratio has changed and there is a higher wattage thermocouple, but I think that is all that is different from the UM2.

TinkerGnome, 2016-03-21:
...and the rotation direction of the feeder is reversed.
Other changes are the additional temperature settings (for each nozzle size). I have "adopted" these for all variants now.

scottdanesi, 2016-03-23:
Awesome!  I will get this installed and tested.

Blizzke, 2016-04-07:
Can you perhaps give a status update about this issue?
I actually received my upgrade kit a couple days ago, but it's still sitting in the box because I refuse to go back to stock firmware for it  :) So I'm kinda waiting on this issue to be resolved.

TinkerGnome, 2016-04-08:
which one of the issues...? :smile: 
[V16.03.1](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V16.03.1) needs testers - have a nice weekend!