TinkerGnome/Ultimaker2Marlin #24: Feature request: adjust Change Filament speed

EldRick, 2016-03-16:
When changing feeders on the UM2-series, it is sometimes necessary to adjust the speed of retraction during a Change Filament operation, such as with the BondTech feeder and perhaps the official UM2 feeder upgrade.
With the BondTech, the retraction is much too fast, and the feeder screams, but a Marlin recompile is needed to change it.
This feature request is for an adjustment to the Change Filament retraction speed, from the display on a UM2.

TinkerGnome, 2016-03-20:
Hi Rick,
the speed is not manually adjustable with the [latest release](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V16.03), but now it is depending on the steps/mm. This works well for the "plus" feeder and should be usable for the bondtech too.
If you want, give it a try and report back.

EldRick, 2016-03-20:
Will do. Thanks! Be sure to let the guys in the Ultimaker Bondtech threads know.

Rick

EldRick, 2016-03-20:
Well, with the BondTech it’s not as fast as the OEM feeder on Change Filament retraction, but at least the BondTech doesn’t scream and stall.
Nice.

Regards,
Rick

closed 2016-07-31 by TinkerGnome