TinkerGnome/Ultimaker2Marlin #26: Hotend cooles down suddenly

Alex9779, 2016-04-10:
After upgrading to 16.03.1 yesterday I tried to print a part I printed earlier with no success because during the first layer the hotend temperature was set to 0.

I am pretty sure that is due to the new firmware because I was on 15.12 the day before and used the exact same gcode file and printed it twice and it worked.

Printing that file with the 16.03.1 had the issue.
Tried it twice, the point where the hotend cooled down was not the same. First time I tried the new full interface during USB printing but didn't change anything. just scrolled around.

Second time I did nothing, just started the print. I almost complete layer 1 but then the hotend temp was set to 0 again.

I tried it printing with USB only... Can't say if it would happen on an SD print too...

Alex9779, 2016-04-11:
Ok Trend again, the same file, with 15.12 and it works again.
Didn't try the first 16 release yet...

TinkerGnome, 2016-04-22:
if you're able to try it again: does it work, if you disable the heater timeout?

Alex9779, 2016-04-22:
Going to test when I have some spare time... Will report...

Alex9779, 2016-04-23:
Ok did again a test with 16.03.1.
To be sure nothing is bad I restored factory settings. Calibrated the bed and the PID.
I didn't change anything, heater timeout was 3 minutes.
I started a print, all seemed fine but after 8 layers the temperature was set to 0 again.
Then I deactivated the heater time out, at the moment the same code file is printing at layer 14 and still fine.

I tried to find a cause but I have nothing special in the file at layer 8, the fans start at 10. Not thermal runaway error...

So it seems related to the heater timeout though the print ran about 20 minutes until layer 8. It was not after 3 minutes of the print...

Ducati848, 2016-05-03:
I encountered this situation with 16.03.1, too.
My machine is Ultimaker 2 extended +, and I am always print with Octoprint.
The hotend cools down after 10mins of print with heater timeout enabled.
After 3 times of failure printing, I found the solution that TinkerGnome provided.
This situation went off after I disabled heater timeout .

MaxWinterstein, 2016-05-22:
have that same issue too. printing via Octoprint and print dies after a few minutes. Tried multiple times. Reverted back to original firmware and same file prints fine. Please fix this.

Alex9779, 2016-05-23:
@MaxWinterstein did you try to turn off the heater timeout? This fixed it for me and Ducati848...

MaxWinterstein, 2016-05-27:
@Alex9779 yes, disabled it and now it is working fine. feels strange somehow...

Alex9779, 2016-05-27:
Yeah, I liked that feature too but I wanted to have the tune options when printing with USB. That is so cool now because I print nearly exclusively with OctoPrint...

sputtering, 2016-07-05:
I was using Tinkergnome build 16.03.1 on a Ultimaker 2+. Same issue, and disable the heater timeout fixed it. What exactly does that setting do? And is this something that might be fixed in the next release?

TinkerGnome, 2016-07-05:
It disables the nozzle heater after a certain amount of time without any user interaction (e.g. if you start a material change and then leave the printer for a while...). Of course it is not intended to happen during a print... That is easy during a print from sd card, because the printer knows its state, but if the printer is connected via USB it's not so obvious...
The firmware assumes that the device is printing, as long as it can read commands from the serial connection. My suspicion is that this detection failes now and then... And yes - someone should fix it... :-)

sputtering, 2016-07-07:
Thank you for that response. Sorry if my comment came out snarky!

ieol, 2016-08-23:
Hi to all!
Fist I wish to say thank you for this nice firmware! 
Second, I never had this kind of problem on my Ultimaker 2 bought at the end of 2015.
Never used USB to print indeed...
But I've downgraded the Firmware to 16.03.1 because with this last release (16.08.1) when the printer end the job, first the head insist on the object for a while (longer than normal compared to the previous firmware) then the feeder motor run again forward and than backward (and more of the setting of 3mm that I have choosed in the settings menu, about 15mm more!)
Worst if you abort the job while printing the first layer! The feeder motor run forward for a while grinding the filament (as the print head is stopped on the build plate!) then the build plate go down and the print head moves away.
So, for me Version 16.03.1 is still flawless

Greetings

TinkerGnome, 2016-08-28:
16.08.1 fixes this issue. Heater timeout can be re-enabled even for Octo-printers. :smiley:

closed 2016-08-28 by TinkerGnome