TinkerGnome/Ultimaker2Marlin #27: "usb communication" shown when in menu

derbroti, 2016-04-18:
Hi,

in the latest release: "for all "Octoprinters": fully functional print menu during usb communication"
This is great! but the usb communication screen also shows up when just looking at the main menu.
It is a little distracting when you currently try to change a setting and octoprint distracts you by splashing the view that looks like I am printing something.

Is it possible to only show the usb communication view when really printing? 
Generally I don't care if USB communication happens - especially when I am in the config menu ;)

TinkerGnome, 2016-04-22:
...it should ignore the M105 commands, but all other gcodes are interpreted as "printing". AFAIK one can see the commands on the Octoprint console view. Are there any other gcodes send (besides of the M105)?

derbroti, 2016-04-22:
Oh, you are right ... hm... octoprint was refreshing the list of files on the SD card (M20). I once had it show up when octoprint restarted - there I have no history in the command list - so I can not currently say what it was doing there...

closed 2016-07-31 by TinkerGnome