TinkerGnome/Ultimaker2Marlin #28: Resume at wrong position?

Blizzke, 2016-04-20:
On my UM2+ (TG v16.03.1) simply pausing and then resuming apparently restarts printing at the wrong position.
Can't measure it exactly, but it seems to start 5mm more towards the front of the printer and 5mm more towards the right (side of the knob). 

[image]

Changed the filament manually: paused, changed filament, resume with the result above.

The time before I used the change material function but somehow that one lowered the Z twice and never raised again and resumed printing in the air afterwards.

TinkerGnome, 2016-04-22:
good point - i rarely use the pause function and never the change material wizard. There's a good chance that i didn't noticed a problem there (yet)... ;-)

TinkerGnome, 2016-04-25:
wait a minute... how did you triggered the pause? With the menu on the printer or via gcode (which gcode?). And was it a print from sdcard or via usb? Each way is different...

Blizzke, 2016-04-26:
Nope nope, a regular ol' pauze via the printer menu (same for the change material). Printing from SD.

JUSTDO1, 2016-05-03:
HI there,
It happens for me as well. i own um2 extendend and the pause (in the printer menu) makes the same problem as shown Blizzke.
Thanks so much for developing this. :+1:

arduware, 2016-05-27:
Hello, with me as well.
When I press pause on the printer, the pressure is continued added.
but does not occur directly after a restart of the printer.

armin-fx, 2016-07-03:
Hello,
I think that is the same cause:
I press pause on the printer.
After the printer is paused I press abort, the printer make strange things.
The head will move anywhere like the print goes on on a different position.

TinkerGnome, 2016-07-31:
It's hard to reproduce, because it does not happen every time.
I tried a fix and made a dozen attepts (with success). Can anyone try it with [this release](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V16.07) to confirm a difference...?

closed 2017-01-01 by TinkerGnome