TinkerGnome/Ultimaker2Marlin #29: Download hex and self build differ significantly

derbroti, 2016-04-20:
Probably just my fault but why does my build hex file differ by 10469 lines - so not just some minor build config things.

I downloaded the provided zip in the release  https://github.com/TinkerGnome/Ultimaker2Marlin/archive/V16.03.1.zip
then build my own with the proviced package.sh script and compare to the also provided *.hex the files are different.

Where is the problem? Do you have to use specific arduino library versions? 
I tried to build with arduino 1.6.6 on a mac and 1.6.8 on ubuntu.

TinkerGnome, 2016-04-22:
as long as it works, i wouldn't care...
The hex-files here are compiled with Arduino 1.6.5. on a 32bit Windows7.

derbroti, 2016-04-22:
In principle you are right "as long as it works" but having different binaries means different potential bugs - which worries me because then people might encounter problems that everyone else can not reproduce because their build system produced different binaries. 

-> version 1.6.5 on windows7 gives me compilation errors :( 

/avr/libraries/Wire/utility/twi.c:364: multiple definition of `__vector_39'
UltiLCD2_low_lib.cpp.o:UltiLCD2_low_lib.cpp:200: first defined here

closed 2016-07-31 by TinkerGnome