TinkerGnome/Ultimaker2Marlin #30: Z motor on E1 instead of Z

korneelb, 2016-05-01:
Hi;
due to a failed stepper driver, my Z motor can no longer be connected to the Z connection., can you create a custom firmware for me where the Z motor connects to E1 on the UM2 mainboard instead of Z? 

this is for an UM2 that has gotten the UM2+ upgrade kit, so UM2+ firmware

armin-fx, 2016-05-06:
Hi,
I think you must modify the file pins.h.
Seach for "Ultiboard v2.0 pin assignment" and copy the numbers from
E0_STEP_PIN  E0_DIR_PIN  E0_ENABLE_PIN
to Z_STEP_PIN  Z_DIR_PIN  Z_ENABLE_PIN.
Note that E1 on board is E0 in firmware and E2 on board is E1 in firmware.

korneelb, 2016-05-06:
So I tried to do that but now acceleration is all messed up.. so I think I did a pretty poor job... I'm trying to see if TinkerGnome would do this for me: )

armin-fx, 2016-05-06:
Ok, that's bad.
I thing you should set the value Z_STEP_PIN, Z_DIR_PIN, Z_ENABLE_PIN to -1 to disable this.
The Z-driver is in another mode (16-step) then the E-driver (8-step).
You must double the Z-value from DEFAULT_AXIS_STEPS_PER_UNIT in file Configuration.h (third entry).
I hat try this on my Ultimaker 2, I had to make a factory reset to update this setting.

korneelb, 2016-05-06:
well,the thing is that the X and Y steppers are affected.. that is the weird thing..

TinkerGnome, 2016-07-31:
sorry for ignoring this... ;-)
There's simply not enough spare time to fulfil such desires...

closed 2016-07-31 by TinkerGnome