TinkerGnome/Ultimaker2Marlin #32: Present for material are not stored

ARKopp, 2016-05-08:
I create a present for a material but it is not stored.

scottdanesi, 2016-05-13:
I have noticed this too.  After flashing to a new version of the firmware the presets are reset back to factory.  There may not be anything that can be done about this.  I simply export my presets to the SD card before the update and then re-import them after the update.

ARKopp, 2016-05-13:
yes but what happened to us that you cant put any new presents they dont
get saved ...

scottdanesi, 2016-05-13:
So you are unable to import presets from the SD card?  Or are you saying that you never stored the presets to the SD card so they are now gone?

ARKopp, 2016-05-13:
i try to make a new present in the menu on the printer and I save but then
they are gone.

scottdanesi, 2016-05-13:
Oh, that makes sense.  Have you tried to manually modify the MATERIAL.TXT file on the SD card root?  You can add new materials here and import them into the machine.  

It sounds like you are encountering an issue creating new materials on the machine though.

scottdanesi, 2016-05-27:
OK, I am now seeing this problem.  Looks like the presets are being reset on my machines as well.  If there is any testing or further detail needed to get this corrected, please let me know.

scottdanesi, 2016-10-25:
Has this issue been addressed at all?  Just curious if this is something that will be fixed in a future version.  Thank you!

gr5, 2017-02-10:
My first thought is that this works fine as long as you reset to factory settings one time.  And every time you upgrade versions of firmware.  But that the way you change a preset is confusing in the gui.   You click something like "customize" and then yo make your changes and then you hit save.  At that point I always think I'm done - but then you have to select where to save your changes.  I had to do this like 5 times before I figured out it wasn't a bug but just an awkward gui.

xile6, 2017-10-31:
Has thos problem been fix in an firmware?
I believe i am on 16.12

gr5, 2017-11-01:
I don't think there are any bugs left related to material settings other than the user interface is a bit confusing.  It's really best though to save your settings to materials.txt and edit that and then load it back in.

xile6, 2017-11-01:
I have them loaded to the materials.txt but every time i import them and power off there lost.
Have to import them back every time before i turn on the printer before printing.

But what i ended up doing was building the firmware with my settings in it.
I will try one of the update firmware later. I have no problems with this one beside that one little thing.

Thank you