TinkerGnome/Ultimaker2Marlin #33: Pause and Change Material Issue

scottdanesi, 2016-05-13:
Hello,
I just upgraded to v16.03.1 from 15.12 on both of my Ultimaker 2's and am now having issues when pausing and changing materials during the print.  When resuming after a material change, the printer gets confused as to where it is and cycles home twice, retracts the filament and tries to continue printing.  This is a major issue as the filament is so far retracted that it will not extrude.

Here is a snippet of code that I inserted right before a layer to pause.

;Added by Scott Danesi 5/10/16 to pause to change Filament
;PAUSE and Change Height
G1 Z50 F1002
G1 X25 Y5 F3000 ; bring extruder to the front and bring the z stage up
M0
G1 E40 F60 ; purge nozzle slowly to prime nozzle
; layer 58, Z = 9.36
; inner perimeter
; outer perimeter
G1 X102.845 Y112.768 F3600
G1 Z9.360 F1002

This worked very well in v15.x, but not in 16.03.1.

Have you seen these issues?  Thank you,
--Scott

TinkerGnome, 2016-07-31:
Hi Scott,
sorry for the delay... - how did you change the material during the M0? Manually?
You move the e-axis after the M0 command in your gcode, but there's no reset to the previous position.
You should use a relative move for purging and restore the position afterwards (with G92 E...)

scottdanesi, 2016-08-01:
Hello,

Here is the section of my code that I added to do the pause and change material

 

;Added by Scott D 5/12/15 to pause to change Filament

;Modified by Scott D 5/13/16 rearranged a few commands

;PAUSE and Change Height

G1 Z50 F1002 ; Raise Z

G1 X25 Y5 F3000 ; bring extruder to the front and out of the way

M0

G1 X25 Y5 F3000 ; reset location

G1 E40 F60 ; purge nozzle slowly

G1 X25 Y100 F3000 ;bring extruder back

G92 E0 ;Reset E to Zero

; layer 61, Z = 9.225

; inner perimeter

; outer perimeter

G1 Z9.225 F1002

G1 X101.007 Y106.142 F6000

;G92 E0

G1 X101.997 Y105.891 E0.0096 F1950

G1 X105.249 Y108.264 E0.0475

Etc

Etc

 

 

Scott Danesi

scott@scottdanesi.com

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2016.0.7690 / Virus Database: 4627/12720 - Release Date: 07/31/16

TinkerGnome, 2016-08-02:
Ahh, it looks more complete now... :smiley: 
The home cycles are suspicious, that should not happen - is this behavior reproduceable with a specific file?
More short questions to track this issue down: are you printing from the sdcard, or via USB?
And i assume the E-axis is in absolute mode (M82 in start-gcode)?

scottdanesi, 2016-08-02:
This behavior is definitely reproducible with my file.  I am using the M82 command to set the e-axis to absolute at the beginning of the file and am printing using the SD card.
--Scott

peteruithoven, 2016-10-07:
I'm not sure what home cycles are, but I noticed it always retracts a bit after the material change is completed. This happens when it's paused, using SD-Card, prepared by Cura & Post processing plugin. 
I already opened an issue in the Ultimaker repo: https://github.com/Ultimaker/Ultimaker2Marlin/issues/117