TinkerGnome/Ultimaker2Marlin #35: Tinker 16.03.1 wrong feeder rate

darkdvd, 2016-07-12:
according the value for UM2.1_jarjar (https://github.com/Ultimaker/UM2.1-Firmware/blob/UM2.1_JarJar/Marlin/Configuration.h)
The default steps value for E is 369 (see configuration.h line 371)
Tinker value use 282 causing under-extrusion using new UM2+ feeder

tourdetour, 2016-07-17:
I think it's the same issue when we hit the "change material" button. It never goes all the way back and I have to put it out manually.

darkdvd, 2016-07-17:
Check your value in "ADVANCED" > "Preferences" > "Motion settings" > "Axis steps/mm" > E???.00

tourdetour, 2016-07-17:
I definitely see E282.00. However, I don't seem to have problems with under-extrusion (or do I ?), but the retraction when changing filament doesn't work to the end. Should I change this value 369 ? I use an Ultimaker 2 upgrade to an Ultimaker 2+ using the extrusion and feeder upgrade.

darkdvd, 2016-07-17:
Change the setting and do another print to see the difference.

tourdetour, 2016-07-17:
Hello, thank's for your help, I tried that. Everything seems a bit bigger, but I need to try with no infill to see if it's better. I used to have a lot of under extrusion using no infill. However, It seems I need more retraction because now there is lines of plastic everywhere (like between the comb structure in the infill)

TinkerGnome, 2016-08-16:
Fixed with 16.07

closed 2016-08-16 by TinkerGnome