TinkerGnome/Ultimaker2Marlin #36: Nozzle temp 0 when starting print

marcvandort, 2016-08-02:
Hello,

I've been using Tinker Marlin for about a week now, and I must say I love it. It includes a lot of features that I missed, and also a bunch I hadn't thought of but can't do without now I now they exist.

I noticed a few times that when I start a print, for some reason the nozzle temperature is set to 0. It doesn't happen every time, and I haven't been able to pin point exactly when it happens, but it seems to occur more often when I print multiple files after another. I will try to trace my steps when it happens again.

If you need any more info, let me know, I'd be happy to help you.

TinkerGnome, 2016-08-16:
Let's try to track this down:
- Which firmware version? Does it still happen with V16.07?
- Are you printing from sdcard or via USB?
- Do you use UltiGCode (default from Cura) or RepRap/Marlin gcode files (from which slicer)?
- If you use UltiGCode - can you check all nozzle temperatures in the stored material profiles (there are 5 temperatures for different nozzle sizes). If in doubt, export the material settings to the sdcard and post the content of the MATERIAL.TXT file here.

marcvandort, 2016-08-16:
I am currently running 16.03.1 I think. When my print is finished I'll look it up for you, I can't seem to access that menu at the moment. I run all my prints from sd card, and I only use Cura. As far as I know I haven't changed any of the default nozzle temperature settings, but I'll check it as well.

I've been noticing this bug quite often. It happens when I print files one after another, but also after I used the move material option (I use this most of the time to change filament, in stead of the regular menu). It happens both during heat up for printing, as well as when using the preheat menu. Even when I manually turn on the preheating for the nozzle, I have to set the temperature manually. Other times it is automatically set to 160 degrees (I think).

Do you want me to upgrade to 16.07 first?

TinkerGnome, 2016-08-16:
The target temperature of the nozzle stays at zero until the temperature of the buildplate is near it's target. As soon as the buildplate is warmed up, the target temperature of the nozzle will be set automatically.
This is intended behavior, is it possible that you noticed just this?

In addtion: for the "Move material" menu the target temperature is only set, if the current temperature is below a certain amount (not if the nozzle is still hot). I forgot the reason for this behavior and this will be changed with the next release.

marcvandort, 2016-08-16:
Hmm, that might have been the case sometimes. However, I did have a print start with a cold nozzle at least 2 times, which is why I started looking for this behaviour in the first place. I'll watch the heating up patiently next time to see what happens.

marcvandort, 2016-09-11:
I have not been printing as much lately because I was on vacation. Also I haven't run into the problem anymore. Unfortunately I can't use TinkerMarlin anymore because I made some upgrades to my um2 that requires some firmware modifications. When I have some spare time I'm definitely going to look into applying these changes to TinkerMarlin, I'm already missing it :)

closed 2017-01-01 by TinkerGnome