TinkerGnome/Ultimaker2Marlin #37: Tinker V16.07 print temperature wrong

nicolinux, 2016-08-08:
With the latest rc (Tinker V16.07) I've noticed that when I start a print and regardless of the filament profile I select, the temperature is always set to 195°. I have to manually set it to the correct temperature via the tune menu. I have performed a factory reset but the problem still persists.

Tested on an Ultimaker 2 Extended with UltiGcode.

TinkerGnome, 2016-08-14:
Huh - no more Octoprint?
Fascinating... :smile:  - where does this value come from? 195° is the default temperature for PLA and 0.25mm nozzle...
I assume that there goes something wrong with the additional temperature settings for the different nozzle sizes.

Two short questions:
- Are your customized material profiles still there? Can you try to export and re-import them?
- Does your gcode contain comments for the nozzle size at the beginning (from newer Cura versions)?

nicolinux, 2016-08-15:
Sure, I still use OctoPrint - just not with the UM2 Extended yet.

I didn't create new material profiles - this was just the printer after a firmware reset. I then did customize the default PLA profile and set the temp above 200° but apparently it was not saved.
I've sliced with the old Cura 15.x. I'll try the new Cura too then.

TinkerGnome, 2016-08-16:
Ok, that means:
- the print starts with the temperature that is stored in the current material profile (so this part is working correct?)
- but the temperature in the material profile was different from what you expected (probably not stored and lost after a restart? - or all was resetted to the default values?)

Or did i understood something wrong now...?

killerkaos, 2017-04-20:
Hi, the temperature has been resolve in the new firmware? I download this one V16.07 and has the same problem. [picture](https://d2py9w124w2itd.cloudfront.net/photo/image/2500x0/58f82c5f7a858/temp_offset.jpg).
I just download the new firmware and have the same problem. The target is 190 and i get a 210 degree. You can see the issue i have in here [link in ultimaker community](https://ultimaker.com/en/community/44261-issues-with-175mm-um2-extended)