TinkerGnome/Ultimaker2Marlin #38: Tinker V16.07 - Buildplate rises after faild print

digifiel, 2016-08-11:
With tinker 16.07 and a UM2+ after canceling a print the build plate goes complete up instead of down.
this causes the heatedbed to crash into the extruder.

TinkerGnome, 2016-08-16:
I made some tests and noticed, that a direct abort from the printing screen works, but if you try to abort a print, that is already in pause state, still strange things happen...
Can you confirm this?

digifiel, 2016-08-17:
I'll try to flash back the firmware this afternoon and will do some testing.
Can't remeber if it happend after a manual abort or with the tunell sensor.

digifiel, 2016-08-17:
Had some spare time this morning so tested it already.

Direct abort without any problems, abort from tunell sensor or pause will cause the buildplate to crash into the extruder.

TinkerGnome, 2016-08-19:
Hi digifiel,
i hope i found the reason - aborting from the paused state seems to work now.
I you want, give it another try with [V16.08](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V16.08)

ieol, 2016-08-23:
Hi, aborting while the printer is running on the first layer still cause the build plate to get pushed towards the print head and the feeder motor run forward fast for a second or two then goes backward longer than the settings (for me settings is 3mm the filament goes backward about 15mm! )

Regards

digifiel, 2016-08-30:
No problems resuming after a pause by the tunell sensor, also it looks that the material flow is better now.

Will test it this week and if i get any strange errors i will let you know.

digifiel, 2016-08-30:
Had some spare time at work this afternoon.

firstlayer pause -> abort will cause the printbed to move down and then complete up again.

Also after flashing the firmware and level the printbed will cause smashing the bed to the extruder.
if you switch of and on there is no problem, so normaly there is no problem but forgot to reboot after the flash.

TinkerGnome, 2016-09-16:
Bug fixed with revised version 16.08.2

closed 2016-09-16 by TinkerGnome