TinkerGnome/Ultimaker2Marlin #41: New feeder is in reverse, old one is running correctly

SubZero1993, 2016-09-24:
Hi,

after the upgrade from Ultimaker 2 to Ultimaker 2+ we got a new extrusion feeder, we have built in the old one as the second feeder for the second noozle. With the latest (Tinker-MarlinUltimaker2-dual-16.08.2) firmware there is the problem that the old feeder ist turning the right way (feeding PLA to the nozzle), but the new shipped feeder is turning in reverse. 
So if we want to print with the first nozzle the Ultimaker is removing the material instead of feeding it to the nozzle.

Is there a possibility to add a function like inverting the signal to the two engines/feeders, so if people have different hardware parts, they can activate/switch/invert the proper signal via the menu on the Ultimaker? (Like switching between primary and secondary nozzle).

Guys, I hope you can help me, your firmware is awesome!

Turtletrumpet, 2016-09-27:
Hello,
this is not a problem of the firmware. My Ultimaker 2 Extended plus runs without reverse with the original feeder! With the Bondtech QR Feeder and the same firmware(Tinker V16.08.2) the filament runs reverse. i changed to pins on ther stepper motor of the bondtech feeder and all works fine.

regards,

trumpet

SubZero1993, 2016-09-27:
Oh that sounds great!

We thought about that too, but we wasn't sure which of the four pins we should switch.. The two inner ones or the outer ones?

Turtletrumpet, 2016-09-27:
Hello,

I don't know, which two of the four cable/pins you have to change, because this is motor dependant.
sry. Please ask the support of ultimaker, they should know that.

regards,

trumpet

SubZero1993, 2016-09-27:
Hi,
sure no problem, thank you a lot for your advice! 
Kind regards,
SubZero

closed 2016-09-27 by SubZero1993