TinkerGnome/Ultimaker2Marlin #42: Manual Temperature Control Below 100 Deg C Not Accurate

scottdanesi, 2016-09-27:
I was doing a cold pull to clean my nozzle on my UM2+ and when I went to set the temp at 90 deg, it would not go that low.  It stayed up at about 110.  I then set it to 70 deg and it hovered around 90, so I was able to proceed.  This happens on both the custom nozzle temp menu and the move material menu.

scottdanesi, 2016-09-27:
All higher temps are accurate.

TinkerGnome, 2016-09-30:
This is related to the PID settings. Try
Kp = 12.00
Ki = 0.75
Kd = 55.00
That works well for me.

scottdanesi, 2016-10-03:
That did the trick.  Thank you.  Should these settings be the default when loading the new firmware?

TinkerGnome, 2016-10-05:
Well, the PID settings are stored in EEPROM and would only change on a factory reset. But you're right, i can place separate defaults for the plus models (with Olsson block), just in case...
It's noted.

closed 2016-10-05 by TinkerGnome

scottdanesi, 2016-10-05:
Excellent.