TinkerGnome/Ultimaker2Marlin #47: LCD reverts to normal post power up screen during prints over an hour - V16.08.2

RetromanIE, 2016-10-20:
This issue was happening on my Ultimaker 2 Extended, and I have now upgraded it to a Extended + and am using the + firmware and getting the same issue

When printing I have the normal Tinker display screen with all the info and option to change speed/temp/flow etc. After an hour or two it reverts to the normal screen you see once the Ultimaker is booted up (with an option to print/change settings etc). The print will continue fine, and I can make changes in the advanced menu to temp/flow/bed temp, but the info screen during the print does not come back until the next print job. Seems to be an intermittent issue as it doesn't always happen.

As I mention this happens on my Ultimaker 2 Extended (which is now an Extended+).

Turtletrumpet, 2016-10-21:
Hello,

I print with Octoprint and i had today a similar problem. After 12 hours of printing the display doesn't go back to "main menu" after finished the printjob. It's the first time with V16.08.2 that this happen.

regards,

turtle

RetromanIE, 2016-10-21:
Yes I also have that issue since this morning (on top of the other issue). After the print I need to power off the printer to gain access to the main menu again. Not using Octoprint though, just regular print from SD.

EldRick, 2016-11-27:
Same as an issue I reported recently. I do not turn the LCD off.

foreachthing, 2016-12-08:
Maybe random. This print (~3 hours), the screen reverted.
A print over 9 hours was fine.

Turtletrumpet, 2017-03-05:
Hello,

I compiled the 16.08.2 for U2E+ with the compiler Arduino V1.8.1. I used this [link](https://ultimaker.com/en/community/38154-how-to-build-marlin-the-way-tinkerghome-and-ultimaker-do-it-if-you-have-windows) to setup the IDE. Since 3 prints with 18 to 21 hour the problem , that  the display doesn't go back to "main menu" after finished the printjob doesn't happen.

regards,

turtle

gr5, 2017-03-05:
@Turtletrumpet I strongly recommend 16.01

Turtletrumpet, 2017-03-05:
@gr5 Thank you for the recommendation! And sorry, that I have misjudged you in the past. My fault.

regards,

turtle