TinkerGnome/Ultimaker2Marlin #48: Very long retraction during spiralize

nicolinux, 2016-10-21:
During a print with spiralize the feeder motor produces a very long and whiny noise. Looks like for no reason there is a very big retraction move which messes up the print. I don't see anything in the .gcode so I suspect the firmware.

[Testcase_dowload](https://www.dropbox.com/s/moj6z59m5ak0ue1/retraction_spiralize_testcase.zip?dl=0)

nicolinux, 2016-10-22:
As a followup - I just printed the very same .gcode file again and the problem didn't occur. I do suspect the firmware even more now :)

TinkerGnome, 2016-10-24:
Perhaps one of the random "printing out of printing area", but for the e-axis this time....?
I think, i will replace the sd-card implementation of the current Arduino compiler with the older version for the next release. The newer Sd2Card library is a bit faster, but the older one seems to be more reliable...

nicolinux, 2016-10-24:
Where should I look for it? Is it accessible through the menu or should I retrieve value?