TinkerGnome/Ultimaker2Marlin #51: two new issues in 16.8.2

EldRick, 2016-10-31:
First, Thanks for your great work on the Ultimaker code.

Two recent issues: 
One is reproducible: when I go to Move Material, I have to click the distance three times to make it work. Once to select, once to deselect, and again to re-select before it will move the material.
The other is occasional: four times recently during a print the display reverts to the initial six-item menu, rather than the print status page. The print continues to completion, but it's annoying to be unable to see status or change anything.

This is on a UM2 with 35w heater, Olssen block, and BondTech feeder.

RetromanIE, 2016-10-31:
I've have the issue with the display on my Ultimaker 2 Extended +.

ieol, 2016-11-01:
Thanks for this fine Firmware!
I tried to reproduce the issue above through the "MOVE" Menu, or through the "ADVANCED-->Move Material" Menu the problem comes out.
No problem on my UM2 (25W Heater, Original Stepper Motors)
Never Experienced the problem mentioned about the display.

There is a way to set the temperature above 300°C ? (I need at least 380 °C but it would be nice if the setting could reach up to 420°C) I have some filament of PEEK to print and the only way is to bypass the power supply and the control board with an external Omron temperature controller
Kindly regards!
Claudio

mechaMecha, 2016-11-19:
I've had the same problems as EldRick. I've had to click three times to Move Material. First click does nothing. I've noticed that it happens on some days, on others it doesn't. And I noticed for the first time today, in the middle of a print, the display reverted back to its normal state (what it shows when it's not printing) instead of showing the print status and adjustment menus.

Thank you for making this firmware available. It's a big improvement over the stock firmware!

TinkerGnome, 2017-01-01:
@EldRick , @RetromanIE , @ieol , @mechaMecha 
Can someone re-check, if the behavior is better with V16.12 (for both issues)? Thanks!

EldRick, 2017-01-08:
I haven't tried a long print, but the Move Material three-click issue is still there. 
If memory serves, on 16.8.2 it did not make the clicky sound if you turned the knob after only one click, but it now does so.

EldRick, 2017-02-03:
The triple-click to get Move Material to work is still there in 16.12.1

TinkerGnome, 2017-02-04:
Ok, it's noted. 😄 
I can confirm this. For me it works sometimes at the first try, but not always... It seems, that the stepper driver will not be enabled for some reason. Let's see...

mechaMecha, 2017-02-04:
Hmm. So I tried out your new firmware for a bit. Haven't had the same problems, but on two occasions, my printer just stopped printing in the middle of print jobs. The display showed an ERROR: STOPPED message (or something like that). It seems like I only started having problems with your firmware after I began changing motor current settings. I apologize, but I'm going to have to switch back to the Ultimaker firmware.