TinkerGnome/Ultimaker2Marlin #53: Sudden ultra-slow movement during multi hour prints

Topy44, 2016-11-22:
On long prints sometimes the printer suddenly slows down extremely, to the point where you can barely see it move. It appears that it is indeed still running, but some glitch caused it to change speed to around .1% of the speed or so. The menu is reactive and the print can be paused or cancelled (though it takes forever to get through the cache). The speed setting remains at 100%. Reprinting the exact same file will sometimes work, sometimes it will produce the same error again at a different place.

This might be related to #22, the error described there has also happened to me once. Both could possibly be caused by erroneus reads from the SD card or corruption of the gcode buffer.

Printer is an Ultimaker 2 with 2.1.13 mainboard. The print is a large object at .1mm layers and 120mm/s print speed, sliced with Cura 2.3.1.

Topy44, 2016-11-27:
I have flashed the current stock firmware back to the machine and it has been running rock stable for several days now. I am pretty sure that it is an issue with the TinkerGnome fork only.

I was thinking what could possibly cause the error and corruption of the GRBL gcode buffer seems the most likely option to me. Not sure how to debug the error though...

nicolinux, 2016-12-18:
I've had the same thing happened for a short 1 hour print. I can't say for sure if the nozzle still moved but it sure stood still for long enough to ruin the layer/print and I had to abort it. Might also be related to #48 so I hope @TinkerGnome will revert to the old sd card code.

Aimeethyst, 2017-02-21:
I've had this same issue occur multiple times now, printer suddenly slows to a crawl, usually a couple of hours into a print.

I always print via USB from a Raspberry Pi running OctoPrint though, so that would suggest that this problem is not SD related?

Dim3nsioneer, 2017-06-13:
To bring this issue a bit further (I have it as well on 16.08): Do you guys experience it after you cancelled another print before? I think this might trigger it here... but not sure.

nicolinux, 2017-06-13:
@Dim3nsioneer Good idea. I got into the habit to power-cycle the printer after every print when I am printing from sd. Using 17.x right now and didn't experience this problem any more.