TinkerGnome/Ultimaker2Marlin #55: Changes to material settings lost when power cycling?

andersolsson, 2017-01-09:
First of all, thank you tinkergnome for this excellent software, it made my printers way much more powerful and pleasant to use!

I am running the Tinker-MarlinUltimaker2go-HBK-dual-16.12.1.hex on my modified UM2Go.
The heated bed requires a bit higher temperature than the default setting in the PLA profile but if I change and store the setting, the changes are lost after power cycling.
Haven't tried to make a custom profile yet.

I had similar issues when I switched between hacked and default firmware long time ago and some of the material profiles had temperature settings that were higher than the maximum setting for the default firmware.

scottdanesi, 2017-01-09:
This has been an issue for a while now.  

https://github.com/TinkerGnome/Ultimaker2Marlin/issues/32

EldRick, 2017-02-10:
Just realized I've seen the same thing, but had overlooked it.

scottdanesi, 2017-02-10:
I would LOVE an update to fix this.  I print with weird materials that need specific settings, and mine get lost every single time I power cycle.