TinkerGnome/Ultimaker2Marlin #56: 16.12.1 unstable

JackalWorks, 2017-01-09:
Prints seem to fail a lot more often with this than 16.08.2. The error message just says Error Stopped go to ultimaker... Not very descriptive, so I am not sure what is causing this. I am on a UM2E+
Now it also hanged while trying to read the list of files from the sd-card.

waumand, 2017-01-15:
Hi, me too : 
with the new firmware (16.12.1) I got the same error "Error Stopped", this happen nearly everytime for long prints (for now, more than 1h). So I'll wait a correction because I double check my feeder, nozzle, temp and all in my 3D printer : I don't find any trouble.
I've an Ultimaker 2+ and a Bondtech extruder in 1.75mm (with 1.75mm Matchless Nozzle), I need the new firmware because I have to reverse the E axe to make the Bondtech working in the right way. With the stock extruder and the previous firmware (16.08.2) I never had this error before.
Please @TinkerGnome find what's going. But indeed, thank's for all your time developing this great potential firmware <3
[image]

TinkerGnome, 2017-01-16:
At the moment I have not more information than - something non-reproducible goes wrong in your case - :worried:
Let's try to find out, what's different: which slicer? version? in case of Cura- any post-processing scripts? Issues only for specific gcode files? A different behavior, if the printer is re-started before every print?

waumand, 2017-01-16:
I use only Simplify3D 3.1.1 (working without any trouble before this new firmware).
I always restart the printer before print, and also here because I need to (after the ERROR - STOPPED, we can't do anything at the screen, but in Octoprint it's working).
I've upload a zip with 2 gcode with the "ERROR - STOPPED" appear randomly, the config for Simplify3D and the .stl file.
[Error Stopped.zip](https://github.com/TinkerGnome/Ultimaker2Marlin/files/707678/Error.Stopped.zip)
I'll try with a new big 3D print file, but this one always bug on the new firmware, so you can try it to see the bug (I hope).

JackalWorks, 2017-01-16:
I also use S3D 3.1.1. I did not try Cura. I downgraded back to 16.08.2. And it is rock stable again.

unixhelden, 2017-01-16:
Tried both, Cura 2.3.1 and 2.4.Beta (and Beta2).

Prints with Octoprint via Serial or SD Card the same behavior. Sometimes the print passes somestimes this error appears, happend with the same gcode, first print success, next try error.

Ultimaker 2+

TinkerGnome, 2017-01-16:
> I downgraded back to 16.08.2. And it is rock stable again.

That's really odd, i have reports that are the exact opposite... Some users had weird issues with 16.08 and are happy again with 16.12... I have nothing changed that could be (directly) related to this. There's no starting point....

JackalWorks, 2017-01-17:
I should point out that I have still other issues with 16.08.2 like the previously reported bugs with the UI like going back to the menu during print and hang in the UI. But when it comes to the actual print it completes successfully and never fails.

gr5, 2017-01-17:
tinkergnome maybe change the path planner lookahead from 16 points to 8 points to see if that helps?  Just for information gathering?

andersolsson, 2017-01-24:
I also had issues (with the Tinker-MarlinUltimaker2go-HBK-dual-16.12.1) on the UM2Go om two occasions.
Two weeks ago I just found it like this after a print:
[image]
It worked fine next time I printed though, same file and same settings.

Today it suddenly started driving the head into the back of the printer in the middle of printing a robot. 
It continued pushing the head towards the back of the printer for maybe 5 seconds, then drove the head into the front of the printer for about one second, and then continued printing in the air at that spot.
I printed the file once again now though and then it worked fine.

I am thinking if it is some scheduling issue that overloads the PSU, or something like what gr5 mentions?

It might be that my case is related to EMI though, since the electrical connection for my printer lacks an earth wire at the moment. (Which can be felt if touching metal parts of the printer)

gr5, 2017-01-24:
Maybe I'll reduce path planner by 2X and send out copies to whomever wants.  Just to collect some info.  Who will test it and what version of hardware do you need me to build?

TinkerGnome, 2017-01-25:
Actually the tinker version does not use more RAM than the standard firmware, but more program memory ofc. I agree, that there is probably a buffer (or integer) overflow somewhere, but I'm afraid, that it's not in an obvious place...
None of these problems were ever reproducible with my machine/settings, there has to be a difference, but where...?
@gr5 Any encouragement is welcome.

gr5, 2017-01-26:
>Actually the tinker version does not use more RAM than the standard firmware
Maybe it only uses more memory on the stack then.  I mean you must have at least one variable somewhere that isn't in the original marlin, right?  These strange crashes make me think "stack" as those errors can be very intermittent and very strange.

derbroti, 2017-01-27:
It has to be something like writing to the wrong ram position, as none of the stop calls that would produce the error-stopped without further messages can not be related to normal printing operation when older firmwares do not trigger anything - ok besides a very erratic safety jumper... (unlikely?)

For what it's worth I build a version for me that is printing the amount of free memory to the unused second USART interface - maybe it runs out of memory after all?

Otherwise happy hunting for bad pointers :|

andersolsson, 2017-01-29:
I just realized one thing when looking at the failed print: It appears like it failed after completing a layer.
So it looks like both incidents on my machine happened when (or very close to) moving the Z-axis.

I have attached pictures and the gcode for the failed robot.
It is not the same gcode as the one where the printer stopped just after printing, that one is stuck on the SD-card at the moment.

Both robots have printed successfully when trying again with identical settings, the first one probably printed fine 30 times or so.
 
Another thing I realized: If it would make any sense, it is possible that my printer behaved strange for about the time of one layer, banging the head into the back and then the front, then starting to move normally again when it reached next layer. (Just based on the duration of the problem).

The first error, when the printer stopped after finishing the robot, was with default settings on the primary extruder by the way
The mid-print issue happened with the second extruder at 370 steps/mm, 1.75 mm diameter and 600mA extruder current.

[UM2G_41_infill_UltimakerRobot.zip](https://github.com/TinkerGnome/Ultimaker2Marlin/files/737517/UM2G_41_infill_UltimakerRobot.zip)
[image]
[image]
[image]

ieol, 2017-02-02:
Beware the SD card. Sometimes seems major problems comes from the SD card.
Yesterday I got a failure printing a new project after 4 hours the print stop, the screen goes black, Led shut off and even heating of bed and hot end goes off I thought was the new Firmware installed a day before  (16.12.1) so I downgraded to 16.08.2 and restarted the print but... the job ended exactly in the same point and with the same problems (all shut off led,heating,display..) I tried to simulate a resume and the printer stop at the same point 74.70 mm.. I took out the sd card, made a scandisk but no problems comes out even if Scandisk sucks I know so... I made a copy of the entire SD Card and.. surprise! not all file were readable and the file I had in print take long to be copied... so after many tries I copied them all and formatted de SD Card (but not fast format, I made a complete instead) Copied again on the SD card and inserted on UM2, my print comes out perfect without any problems made a second print of 8 hours and NO problems comes out
So... take care that the SD card is in good condition before attribute any failure to the firmware 
Cheers to all and Thank again for this nice Firmare

derbroti, 2017-02-03:
I can second the fact that a faulty SD card can lead to arbitrary head movements - but those errors would not result in a "error - stopped" message on the display?

unixhelden, 2017-02-03:
And this error also happens when printing via OctoPrint (USB), so this is not SD Card related.

Turtletrumpet, 2017-02-03:
I print only with octoprint and get the 'stop' error, too. Is it a good idea to make a factory reset, after flashed the V16.12.1? The older version i used is V16.08.2.

ieol, 2017-02-03:
@derbroti a faulty or better a semi faulty or we should talk about a CRC RANDOM faulty SD card can lead at any kind of problem even a corruption of other data depending on the system with which it will interact (software, hardware combinations) it's not a simple problem when you got Random CRC errors (as the case of my SD card)
When I tried to read it trough a 7 doors USB HUB it was merely unreadable, so, I connected it to the frontal USB door and it go back to be readable (well with some retry during copy process)

@unixhelden @Turtletrumpet  Turtletrumpet indirectly gave the answer to you as the Octoprint have REFRESHABLE memory inside, this mean the same, (alike, similar) chip of a SD card... But take in account that SD card born to be erasable a number of time that is usually Bigger compared to a Flash Memory chip used for firmware (you can find more info on the datasheet of the flashmemory chip manufacturer)

BY THE WAY: I believe that TinkerGnome got really a lot of patience reading so many comments...
**SO thank you again for this Nice Firmware**

ieol, 2017-02-03:
@andersolsson I see you comment now (posted 10 days ago) This is right what happened to my UM2 (the print head above the printing piece, all shut off except the head fan that it is not controlled by sofware (as it goes ON when you power the UM) and I am almost sure that it is an SD card problem. I suggest to you to copy all your files on a computer and format your SD card (don't make a fast format as it doesn't correct the problem)

unixhelden, 2017-02-03:
@ieol nope, my octoprint stores and loads gcode and timelapses to snd from a nfs share.
BTW, i never had any hickups with 16.08.2 or the original Ultimaker Firmware.

With 16.12.1 i get spontanous Errors like the one described above. Print stops, Nozzle in the Middle of the Print and Display states. "Error - Stopped". So i would rule out SD Card Problems. Same Gcode File runs perfect and the next Print is an Error.

pi@umocto:~/.octoprint $ grep 'mnt' config.yaml
  timelapse: /mnt/syno/Timelapse
  timelapse_tmp: /mnt/syno/Timelapse/UMtmp
  uploads: /mnt/syno/Uploads
  watched: /mnt/syno/Uploads

192.168.200.250:/volume1/OctoPIUploads/Ultimaker on /mnt/syno/Uploads type nfs 
192.168.200.250:/volume1/Timelapse on /mnt/syno/Timelapse type nfs 

I have 3 Octoprint Servers for 3 different Printers, all setup the same with this nfs share. And only this Firmware brings errors. If someone would provide a debug version of that firmware, i could make some tests with serial logging enabled.

gr5, 2017-02-03:
@unixhelden - I have the build environment setup the same way UM does and tinkerGnome does.  I'd be happy to build a version for you but I don't know about this "debug version".  Is there a flag in configuration.h (sorry too lazy to check at t he moment).  If there is such a flag and then tell me about it and tell me what configuration I should build for (there's a dozen or so builds) and I'll make you one.

ieol, 2017-02-03:
@unixhelden so let resume you use an even more susceptible connection mode for transferring gcode?  Really! seems right the correct way to test something new! And see if it is stable and error free...  gosh!

gr5, 2017-02-03:
The fact that it works on 3 printers for years and "suddenly" stops working when simply upgrading the firmware is good enough for me.  It could be a coincidence with something else that happened but considering all the complaints I think there is a bug in this version of the code.  Probably not Tinker's bug - probably some bug that is only now showing up.  Maybe the interrupt service routine doesn't finish in time.  Maybe there is a stack error or initialization error or... something.

ieol, 2017-02-03:
@gr5 there are no fact or the problem would already be solved... if you think that change in firmware is a simply upgrade you already write all... have a nice day

unixhelden, 2017-02-03:
@ieol please. calm down.

i will summarize that for you: Multiple People experience the same issue when using 16.12.1. Multiple People say this error occurs with different slicers ( i personally tried s3d and cura (2.3 and 2.4beta1-2)).

All of them Report when they switch back to the older firmware this error does simply not happen at all. With the same hardware, the same slicer, the same sd card and or serial delivery via Octoprint. 

All of them say: We can print the same Gcode 2 Times, the first time it's ok, the second time the same gcode on the same sd-card or the same pipeline brings this error up.

Next i tell you from my point there is no sd card at all involved. I use this setup on THREE Printers without any issues, i run prints for my hub which are 36-72 hours long without any hickup or fault unless i run 16.12.1 on my um2+. 

@gr5 i'll come back to you. I have to run several orders till thursday on my um but i think there is a debug option allready in marlin, i will look into it.

Turtletrumpet, 2017-02-03:
Clear & short: There is a problem in 16.12.1. In that Version the pint stops without useable error message. Nevertheless i love the Tinker"ware" ;). I hope the error will be found. I don't know, what i could do, to find the error ...
**@ieol** take a deep breath, be friendly and have a nice weekend ;)

ieol, 2017-02-03:
@unixhelden @Turtletrumpet LOL
I never write 16.12.1 is flawless, I just point out the way you are all begging for a flawless firmware but just write down information that ARE INCONSISTENT for who is developing it!
(UPPERCASE is just for underlining not shouting... I prefer instead of **uppercase**)

oh @Turtletrumpet have a nice weekend too ;)

Turtletrumpet, 2017-02-03:
@ieol please start first and write consistent messages, who helps to find the error. At the moment you keep writing, that our feedbacks ARE INCONSISTENT. My opinion: wrong way to solve problems.

Turtletrumpet, 2017-02-03:
the art of programming is to ask the right questions at the right moment and made a progam of that).
Just after the second world war, when i start programming, my programms stops with debug information ;).

ieol, 2017-02-03:
@Turtletrumpet **LOL** are you silly? or too tired?

Turtletrumpet, 2017-02-03:
@ieol ask this question first yourself ....

gr5, 2017-02-03:
When reading a comment that appears angry at first one should assume the writer was laughing or smiling when he wrote it and was trying to be friendly.  There's usually at least 2 ways to read every post - always assume the friendlier version as that is usually the truth anyway.  I suspect ieol was not angry when he made any of these posts and didn't mean to hurt anyone's feelings.  @ieol I suggest you fill all of your posts with smiley faces in the future :)  All caps doesn't always mean shouting.  It could be mild emphasis. :)  Of course there is a third possibility - that you are dealing with a fool and fools should be ignored.  But always assume the best possibility!

TinkerGnome, 2017-02-03:
back to topic... some thoughts:

- for 16.03 and 16.08 a more recent version of the Arduino sd-card library was used and multiple users had problems with read errors from the sd-card
- at least two of them solved this by re-routing and/or swapping of the two ribbon cables between the mainboard and the controller board at the front of the printer
- to avoid this issue i (tentatively) switched back to the former sd-card library and got no new reports about "printing out of print area"
- now with 16.12 some other users have different issues, while for all others (including me) all firmware versions are working just fine
- the new problem is unrelated to the sd-card, it doesn't matter how the printer is feeded with gcodes
- the un-answered question is: where is the difference? hardware? differences between the mainboard revisions? interfering signals (because the buildplate heater is in PID mode)? user workflow? weak connections? overheating (do we ask too much of the poor old Arduino)?
- ok, this are multiple questions... 😃 

AFAIK the OLED display (and the encoder wheel) is also connected via the ribbon cables, communicates with the mainboard through I2C protocol and the whole process is handled inside an interrupt service routine. Can a bad signal on the I2C bus screw up the Arduino in a way that it reads bad values from input pins anywhere else? I really don't know.

@gr5 - it can still be a stack overflow, but i doubt it. This should be reproducible, shouldn't it? If the printer is powered on, the memory state is always the same. If I print the same file again, it should fail at the very same point. I usually start several prints in a row, long prints, short prints again and again and the printer is running all the time. If there where a serious memory overflow - I'm sure that I would have seen the problem too...

> the art of programming is to ask the right questions at the right moment

So the question is - "what is the right question"... 😃

gr5, 2017-02-03:
One thought - in "normal" Marlin there aren't quite so many things being updated on the screen.  Maybe if you update one thing too many then it takes a bit too long and some timing problem occurs - like on windows when one process is using too much cpu time.  I haven't looked at the display code but maybe the I2C ISR is just pumping out data from some buffer to the screen.  Maybe the buffer for that is overflowing?  Is there a buffer for that?  Maybe the ISR routine takes a different amount of time and sometimes it takes too long?  Maybe the difference is some subtle update - maybe a test is to display FEWER things on the "printing" screen - or to go into the tune menu and leave it there and see if it still crashes?  Does this even make sense?  I don't know without looking at the code.

derbroti, 2017-02-03:
@gr5 but from 16.08 to 16.12 the display functions did not increase in amount, right?

When comparing the the releases, there are not many noticeable differences (when ignoring the SD reader change) but one major looking switcharoo and change is in stepper.cpp in the ISR 
`// "The Stepper Driver Interrupt" [...]
ISR(TIMER1_COMPA_vect) [...]`

in Marlin_main.cpp there are quite a few memsets which make the main loop a little slower but that shouldn't be a problem...

It nevertheless can still be the rolled back SD library - maybe it has a weird bug in it? 
@TinkerGnome I didn't look into it: where does the "old" (the one you are using now) and "new" (the previously used one) come from?

@all are you people using the prebuild *.hex files from tinker or are you compiling yourselves? 
-> If so, which Arduino IDE version are you using?

gr5, 2017-02-04:
I know that TG uses arduino compiler 1.6.10.  TG helped me set up the build environment similar to his and similar to how Ultimaker does the build.  I took lots of notes if you want my notes.  I was going to post them on the forum but forgot so I just did it a minute ago:
https://ultimaker.com/en/community/38154-how-to-build-marlin-the-way-tinkerghome-and-ultimaker-do-it-if-you-have-windows

gr5, 2017-02-04:
I'm happy to build someone a version with the bang-bang heated bed if they want to test to see if that was causing their problem.  It would be nice to start eliminating things.  You could just air-print with a cold nozzle.  Although it might take all 4 axes moving to cause the crash.

derbroti, 2017-02-04:
@gr5 Interesting, a few month ago he was using 1.6.5 and the package script to this day still defines the arduino env as 165
I haven't looked into how much of the default arduino lib stuff is used in the firmware but it is certainly a point that changed

gr5, 2017-02-04:
The lights shouldn't flicker "when printing" but when "up to temperature".  The PID controller says "power at 100%" when it's 10C cooler than goal temperature.  But when it gets within 10C then the lights start to flicker.

gr5, 2017-02-04:
On December 30, 2016 TG said he was using 1.6.10 in an email to me.

derbroti, 2017-02-04:
No, mine also flicker (but almost not noticably) after heating up and at constant temperatures

gr5, 2017-02-04:
Okay maybe I found something.  This line of code was pulled out on december 26 from stepper.cpp (last line in code chunk below).  TG - I don't understand this exactly but I think setting busy=true may be important.  This isr is like a thread and can happen at any moment and the motion planner needs to know that this block is now being sent to the steppers so the planner doesn't change it mid-stride.  Right?  Could this be it?

     // Anything in the buffer?
     current_block = plan_get_current_block();
     if (current_block != NULL) {
-      current_block->busy = true;

derbroti, 2017-02-04:
@gr5 Sadly no the block that is returned by `plan_get_current_block();` is set to `block->busy = true;` automatically (see [planer.h:133)](https://github.com/TinkerGnome/Ultimaker2Marlin/blob/geek_mode/Marlin/planner.h#L133)

TinkerGnome, 2017-02-04:
If anyone is brave enough 😄 to try different compiler versions - the instructions from gr5 are great - it's not that complicated. Any version from 1.0.3 on should be able to compile the code.
BTW: Yesterday I installed on my machine  a version that was compiled with Arduino 1.8.1. There's no noticeable difference in behavior so far. But who knows... if you want to rule this out - give it a try!

qwerty8224, 2017-02-06:
Sura 15.04.6 3 days printing no errors

derbroti, 2017-02-06:
@qwerty8224 Which firmware? Which printer?

qwerty8224, 2017-02-06:
Tinker-MarlinUltimaker2-dual-16.12.1.hex
UM2

rolfx, 2017-02-11:
"Tried printing out of printing area"...

from november 2016 - after mounting the bondtech feeder on my UM II ext - I used Tinker-MarlinUltimaker2-16.08.2.hex. problem-free.
building an object of z-height 300 3 days ago the UM stopps at Z 230mm with the error "tried printing out of printing area." 
using youngest firmware "Tinker-MarlinUltimaker2extended-16.12.1.hex" the result is the same: UM stopps ptnting at Z 230mm.
on an tipp from martin bondeus - bondtech owner - I checked the UM display and find adjust z-position. the nozzle goes to Z 230mm and prompt me to move the platform to the nozzle. one can move the platform about 50 mm - then it stopps.
finding today your discussions over the issues #56 I find one sentence that I do not understand: "to avoid this issue i (tentatively) switched back to the former sd-card library and got no new reports about "printing out of print area".
can _I_ switch back?    to what?? 
I am not coder at all!
and: which is _the_ stable version to build parts up to the max height of the UM II xt?

one sentence to the theme "error bits at copying to SDcard". as I know all "write and read"-actions are verified by the system(s) by "Check totals". otherwise we would have lots of mistakes in pictures and movies on/from SDcards.

thanks for your patience

TinkerGnome, 2017-02-11:
@rolfx - Your problem is perhaps easier to solve. It sounds like you had installed a wrong variant before (not the "extended"). You should check the stored maximum z-position:
Advanced -> Preferences -> Print area
Change the max. Z position to 325.0 and re-adjust the z-position afterwards. This should do the trick (in your case).

rolfx, 2017-02-13:
indeed - good tipp. it works. on the other hand: I wrote and I installed the _extended_ firmware - at least the one that i declared as it. 
thanks!

rolfx, 2017-02-13:
one questions: is there a documentation about all menu-points of the tinker-marlin firmware?

gr5, 2017-02-13:
You mean you want like a visual tree view of all the menus?

rolfx, 2017-02-14:
yes - it seems easier to me to get an overview - and a print in the drawer ;-)

TinkerGnome, 2017-02-19:
Meanwhile I have a strong suspicion about the reason for instability, summarized in: short voltage drops due to an overloaded power supply...
Thanks to @gr5 you can check this theory now. I published a new (testing) release right now with his smart power limiter.
If you have one of the issues described earlier: feel free to test [V17.02.1](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V17.02.1). Please use the default values for the power budget in the beginning and report back, if it improves the stability.
👍 ❓ 👎

derbroti, 2017-02-20:
Did anyone measure that? My printer stays at comfortable 100W during print.

The only possibility: When you try to print abs in a freezing/windy room with >=100C bed temp and a >=40W heater. 

Only during heatup of bed and nozzle (50W heater limited to ~30W) I do indeed reach around 220W (Input power not output)

EldRick, 2017-02-20:
I completed an 18-hour print with no error-stopped, using 17.02.1 This on UM2, Bondtech, 35w heater.

gr5, 2017-02-21:
But you never had a problem before either, right?

EldRick, 2017-02-21:
Not before 16.2.1

qwerty8224, 2017-02-26:
7 days no printing errors )))

[39 more replies]