TinkerGnome/Ultimaker2Marlin #59: UM2+ Extended getting ER07 on home head with 16.12.1

markusmaker, 2017-01-25:
Hi there!
I immediately get a "ER07 Error Stopped X or Y switch broken". With original firmware everything works fine. Its a UM2 Extended which has been upgraded to Plus (by myself). See videos attached.

Before the printer was upgraded I used a older TinkerFirmware on it. I think it was 15.10.x.

I also tried 16.03.1 but same issue.

[videos.zip](https://github.com/TinkerGnome/Ultimaker2Marlin/files/730799/videos.zip)

gr5, 2017-01-26:
Looking at the videos it looks like without changing any hardware you can get the head to home just fine with Ultimaker Marlin but not with tinkerMarlin?  Is that correct?  If there was any (even tiny) hardware changes please also try heating up nozzle.  Homing head and heating nozzle both require 24V internal supply but the display does not.

gr5, 2017-01-26:
Oh - and make sure the printer is turned on!  I think the display will be on even if powered only with USB (not sure I ever tried it).

markusmaker, 2017-01-26:
Hi gr5, tinkergnome told me on to make a factory reset - which worked.

https://ultimaker.com/en/community/37214-tinker-firmware-um2-extended-getting-er07-on-home-head-with-16121

Probably this info with overwriting settings and doing a factory reset after firmware change should be added somewhere.

Thanks.

gr5, 2017-01-26:
Needed factory reset.  Issue closed.

commit d175316d (2015-01-20): So this should fix #59
closed 2017-01-26 by gr5