TinkerGnome/Ultimaker2Marlin #61: 17.02 dev

qwerty8224, 2017-02-13:
does not work UltiGcode

gr5, 2017-02-13:
I was able to print an ultigcode print last night with 17.02 no problem.  But I had sliced it cura 15.X.

What was the symptom exactly?

qwerty8224, 2017-02-13:
Heated bed nozzle. But printing does not start
Reprep/Marlin  working
Later, I can check on the 15.X

gr5, 2017-02-13:
Are you printing through usb or using the SD card?  From SD card use ultigcode.  If printing through USB (which isn't officially supported but works usually) then use reprap mode.

qwerty8224, 2017-02-13:
From SD card use ultigcode

TinkerGnome, 2017-02-18:
Hi @qwerty8224,
how have you compiled your version, and for which printer?
Perhaps some of the compiler defines from package.sh are important?

qwerty8224, 2017-02-19:
Arduino ide 1.6.6
Ultimaker 2
2 extrusion runs through ultigcode Cura 2.3.1

qwerty8224, 2017-02-20:
17.02.1 very good )

closed 2017-02-20 by qwerty8224