TinkerGnome/Ultimaker2Marlin #64: It does not work on UltiGtsode

qwerty8224, 2017-02-20:
Tinker-MarlinUltimaker2-dual-17.02.1.hex
Tinker-MarlinUltimaker2plus-dual-17.02.1.hex

UltiGcode work Tinker-MarlinUltimaker2plus-17.02.1.hex

gr5, 2017-02-21:
Can you please provide more information.  What doesn't work?  And please don't create many issues all about the same problem.

qwerty8224, 2017-02-21:
firmware dual extrusion no work UltiGcode
firmware dual extrusion  work Reprap/marlin

gr5, 2017-02-21:
What do you mean "no work".  Does the printer boot up?  Do the menus work?  Do you print usb or SD card?  Does it work for a few minutes?  Does it crash?  Does the bed heat up?  Does the screen not even come on?  Does it show version upon boot?

What do you mean "no work"

qwerty8224, 2017-02-21:
SD card. Warm Bed nozzle but printing does not start

gr5, 2017-02-21:
This is useful.  Is the display locked up?  Or can you go into tune and abort the print?  Can you post the gcodes here from your SD card up to and including the first movement (first G0 or G1 command) please?

qwerty8224, 2017-02-21:
I can turning display print runs. Gtsode can lay out tomorrow ....

qwerty8224, 2017-02-22:
;FLAVOR:UltiGCode
;TIME:704
;MATERIAL:1028
;MATERIAL2:0

;Layer count: 71
;LAYER:0
M107
G0 F12000 X93.028 Y93.028 Z0.260
;TYPE:SKIRT
G1 F1800 X93.881 Y92.329 E0.11469
G1 X94.849 Y91.812 E0.22882
G1 X95.901 Y91.492 E0.34318
G1 X96.820 Y91.400 E0.43924
G1 X113.180 Y91.400 E2.14068
G1 X114.099 Y91.492 E2.23673
G1 X115.151 Y91.812 E2.35109
G1 X116.119 Y92.329 E2.46522
G1 X116.972 Y93.028 E2.57991
G1 X117.671 Y93.881 E2.69460
G1 X118.188 Y94.849 E2.80873
G1 X118.508 Y95.901 E2.92309
G1 X118.600 Y96.820 E3.01915
G1 X118.600 Y113.180 E4.72059
G1 X118.508 Y114.099 E4.81664

gr5, 2017-02-22:
Good info.  So it warms the bed.  Warms the nozzle.  And then just gets stuck - it never does even the first G1.  But you can go in and choose "abort".

qwerty8224, 2017-03-05:
Yes

Aleksys007, 2017-03-20:
I confirm, the printing does not even begin. 100% appears on the screen and nothing happens.

qwerty8224, 2017-04-07:
This occurs when passing from RepRap to UltiGcode 2