TinkerGnome/Ultimaker2Marlin #65: 17.02.1 Firmware "Error - Stopped" and another potential problem. Ultimaker 2 Extended +

RetromanIE, 2017-03-01:
I've had a couple of things happen since installing 17.02.1 on my Ultimaker 2 Extended +
First issue i am not sure if its firmware. I've had the printer just jam up and start making this terrible grinding noise during some prints(not grinding material or the print, the xy axis just goes crazy) . It can also happen when i click to just change material. It sounds like the printer is tearing itself apart, but once i power cycle it is fine.

The other issue is a common one on the older tinker firmwares, "Error - Stopped" after a few hours printing. See attached image. 

Anyway, I've installed the latest official firmware and reset to factory defaults and all seems well. I much prefer the tinker firmware

[image]

gr5, 2017-03-01:
Yikes.  Okay, thanks for this report.

PilotdG, 2017-03-11:
I have exactly the same issues with my 2+, occasionaly making  lot of noise (like it homes without finding the X or Y switch, ....)
And, more serious, this error -stopped after about 20 hours in a print. 
I love this firmware, but at this moment, it is not usebale on my machine.
With the default fw, everything runs without a glitch.
I have the same error-stopped with the 16 FW.

gr5, 2017-03-11:
Which 16 firmware? I've been using 16.01 for about a year now with no problems.

PilotdG, 2017-03-12:
I tried 16.12.1
So you think 16.01 is more stable?

gr5, 2017-03-13:
Lots of complaints about 16.12.1 crashing.  None about 16.01 that I know of.  The releases page was change about a month ago to recommend to use 16.01.

PilotdG, 2017-03-13:
i tried 16.01, got all kinds of errors, first broken x-y endstop, then the extruder turned backwards, then the display went black, after waking it up again, i was in the original menu.
Put the last officail FW on it again, zo i can get on with my work, until i find the time to experiment again. I LOVE the features of the thinkerware, but at this moment can't get it to work.

gr5, 2017-03-15:
After you install 16.01 you have to do "reset to factory settings".  This is because a lot of these settings such as extruder steps/mm are stored in some kind of prom on the arduino.  Every time you change versions of firmware the firmware makes an attempt to see what older version of factory settings you have and tries to update them to the new way.  But since 16.01 is an *older* version it fails this miserably.  So just reset to factory settings.  Unfortunately it will make you level the build plate but that shouldn't take long.

PilotdG, 2017-03-16:
First, many thanks for your cooperation, appreciate that!
Tried it again, version 16.01 UM2.
Did a full factory reset. X-Y endstops ok now! But, Still the same problem with the  extruder turning backwards. 
I have a upgraded UM2 to UM2+, think the problem will be that the feeder changed.
Is there a 16.01 for the UM2+?

gr5, 2017-03-16:
Oops. Sorry, no.  I have bondtech extruder on my um2.

PilotdG, 2017-03-18:
I tried the 16.03.1, oldest version for um2+, and that seems to work!! I am in a 16 hr job, and everything is working as it should! 
Things is already appreciate, Led and display sleep, babystepping Z, and 270 degrees printtemperature for the PC-plus filament.
GR5, Many thanks for your cooperation, made me find the right version ;-)

Haldahir, 2017-04-24:
Hi,

I have an UM2 upgrade to UM2+ but without chnaging the extruder stepper. The custom firmware allow me to double the number of step / mm. 

I have the same problem. Sometimes I had an error without explanation...

gr5, 2017-04-24:
What version?  If you go to the releases page the top version is "testing".  The next version down says "unstable" and recommends 16.01.  I recommend 16.01.

Aim23, 2017-11-06:
Does this issue still persist with the new Version 17.10.1? 

I'm currently not able to update my UM2E+ for testing (heavy christmas using ;))

gr5, 2017-11-06:
This issue is still not understood.  It could just be a hardware issue for certain users.  I've never gotten these problems with my version which is 17.something.

Ando120, 2018-01-10:
Stopped after about 20 hours for me too with UM2+. 
I lost 5 prints with testings (reinstalled, factory reset, different cura version) 
I turned off Light timer and screen timer and looks fine now. Two 48+ hours prints after that.
Maybe it's related?

Anyway, its very good firmware.

gr5, 2018-01-11:
What is the light timer and screen timer?

gr5, 2018-01-15:
lol.  No I meant what were you talking about.  I'm not aware of this thing.  

After reading your reply I'm guessing it is a tinkerMarlin feature that is like a screen saver maybe?  I don't know all the features.  I wrote the power budget feature but there are features I haven't noticed yet.

Turning off the screen or disabling geek mode is my current guess as to how to avoid these crashes.  There's a lot going on to update all those things in geek mode while printing and my theory is that disabling geek mode results in more cpu time to do critical printing things like motion planning.

Loki44, 2022-07-16:
Still having the "Error - stopped" during a long print problem in 19.03.1

gr5, 2022-07-16:
I'm not seeing these on my 3 um2go's with 19.03.1.  Which I use almost daily.

I'm wondering if this is a hardware issue with reading the SD card.  I've seen problems on older UM2 printers where it gets random flipped bits when reading the SD card.  All SD cards.  I think once every 3 years you need to remove the SD card reader circuit board and clean it out with both vacuum and some solvent.  The symptoms are:
- It will change (for example) the E value for only a single gcode such that it will retract 100mm or 1000mm and then on the next gcode it unretracts - the printer will seem to stop for quite a few seconds while it does the retract and unretract and sit there relatively silently (but the extruder motor is moving) and then continue like nothing happen.  Some describe it as moving extremely slowly because the X and Y axis usually move synchronously but slowly as they are honoring the speed of the E value
- it will change say an X value such that the X axis moves far outside the part - usually a move of 10mm, 20mm, 40mm, 80mm (these are binary changes to the ascii characters - always just one bit) and the move back into the part and continue where it left off - giving you strange bumps/exterior strings on the side of your part
- or the gcode will be so bizarre it might give you the "error stopped" message?  Not sure
- or the extruder will move *forwards* (versus retract) by 100mm and grind up the filament followed by a huge retraction causing massive underextrusion or complete failure of the extruder's ability to move the filament as it's chewed up the filament
- or the X value will move so far you get an "out of bounds" (I forget the exact warning) error on a gcode and it pauses/stops printing at that bad gcode
I've seen all these errors on 2 different um2 printers except "error stopped" and they were fixed by cleaning the SD card reader