TinkerGnome/Ultimaker2Marlin #66: Random Print errors on version 16.12.1e

boxedfish, 2017-03-02:
Hi,

The firmware is really nice BTW and has helped no end with setting up my new extruder current and steps but I have noticed that randomly either after a few hours on one print or if I quickly fire up a second print it randomly throws a ERROR - STOPPED message and the print has stopped with the bed up and the print head frozen in place like the memory card was pulled out. I have tried Cura 15.04.6 and Cura 2.4.0 with various STL files and there is no common pattern.

It may well not be the firmware as I have upgraded the heater, the extruder to a bontech extruder, hot end and a new temp sensor so there is the possibility of it being one of them giving me issues but its very very odd as normally there is some detail as to whats gone wrong but every time the message has been just ERROR - STOPPED.

I have tried slow printing, fast printing, higher heat, lower heat, current ratings from 1300ma down to 1000am on all motor's to printing feet for better airflow just in case the Z axis stepper was causing to much heat on the original 1300ma setting that was on that motor.

But with all the changes I still get the issue randomly.

Any advice would be good or I am tempted to start uninstalling the new toys. But I am 99% sure it did it a couple of times to me before they where installed so might not be that.

Oh and its an Ultimaker 2 Extended

Thanks,

Alex

[image]
[image]
[image]
[image]

gr5, 2017-03-02:
This is a duplicate of issue #56 

Please downgrade to version 16.1 -- see release page: https://github.com/TinkerGnome/Ultimaker2Marlin/releases

Version 17.X is not necessarily better than 16.1.  I recommend 16.1.

Note that a bad SD card can cause this so if you have another SD card try that.  I've gotten "error stopped" because of a bad SD card I think.

closed 2017-03-02 by gr5

boxedfish, 2017-03-02:
Hey @gr5 sorry, yeh just seen its a duplicate issue, sorry, I blame the cider ;-)

I will downgrade as I have tried 4 cards one being an extreme card which I know to be good as gold and only a few weeks old. Thanks for the advice, hope you guys get to the bottom of the random issues, if I could help I would.

Thanks,

Alex

gr5, 2017-03-02:
No problem.