TinkerGnome/Ultimaker2Marlin #67: 17.02 stopped and underextrusion problem

ARKopp, 2017-03-05:
with the latest version i got underextrusion while with an older version I do not. Also a thing stopped a little bit before finishing.

gr5, 2017-03-06:
The underextrusion issue - maybe your steps/mm is off.  Check that - you can get to that through the memory system.  Or recalibrate.

having a print stop before finishing is much more serious.  Have you ever had this issue before?  It could be power supply brick rebooted the machine or it could be a software bug.

gr5, 2017-03-06:
Just to clarify - I'm 99% sure there are no underextrusion related bugs in 17.02.  Either steps/mm is off or something just coincidentally happened unrelated to the firmware.  Maybe look at my list of causes for underextrusion here but definitely use the move menu to move filament 10mm and see if it is actually moving for example 5mm: (my march 5 post here) https://ultimaker.com/en/community/39216-problems-after-several-layers#reply-173903

ARKopp, 2017-03-06:
> after-several-layers#reply-173903

ok I will try. also another questions. where to I find the
startcodes, pause and end codes in the firmeware. I want to put it in
octoprint.

TinkerGnome, 2017-03-06:
I think, it's not transferable that easy - at least not one-to-one. The sequences are spread in UltiLCD2_menu_print.cpp.
It starts [here](https://github.com/TinkerGnome/Ultimaker2Marlin/blob/geek_mode/Marlin/UltiLCD2_menu_print.cpp#L542) after a  file is selected, goes through the [heatup menu](https://github.com/TinkerGnome/Ultimaker2Marlin/blob/geek_mode/Marlin/UltiLCD2_menu_print.cpp#L642) and [primes the nozzle in doStartPrint](https://github.com/TinkerGnome/Ultimaker2Marlin/blob/geek_mode/Marlin/UltiLCD2_menu_print.cpp#L136).

The end-sequence is inside "[abortPrint](https://github.com/TinkerGnome/Ultimaker2Marlin/blob/geek_mode/Marlin/UltiLCD2_menu_print.cpp#L45)". Nothing fancy there, basically: switch off all heaters and home axis.
And the pause command is generated in [lcd_print_pause](https://github.com/TinkerGnome/Ultimaker2Marlin/blob/geek_mode/Marlin/UltiLCD2_menu_print.cpp#L1242). I'm not sure, if the M601 command can be used from Octoprint.

gr5, 2017-03-06:
If you slice something with cura in "reprap" mode (in cura 15.X set machine to reprap mode) it will generate some gcodes at the top and bottom.  You can use those.  Actually I think cura will have a start/end gcodes tab.  So you don't even have to slice - just change your machine settings to reprap mode.

3Dimensionen, 2017-04-09:
Since we updated to the newest version we have severe underextrusion although we checked every possible option. We have multiple Ultimaker² running and all of them are producing underextrusion. So i will switch back to an older version where we had no markable issues.

gr5, 2017-04-09:
I recommend a factory reset.  Some of the parameters stored in eeprom get messed up when you change versions as they are sometimes stored in a different location in the eeprom.  Especially when you switch from a newer to an older version.  Factory reset fixes this.

3Dimensionen, 2017-04-09:
Okay.
Thank you for your quick response. I will give it a try and will report on this issue.

EldRick, 2017-05-15:
I also had underextrusion and backed out 17.2.1. I'll reinstall and check the esteps.

Ricky1975, 2019-01-29:
Has there been any solution? Because I face the same issue with the UM2 since this version.
I did a factory Reset with the M502m checked e-Steps and everything else the same way if it was a new printer.
The other point I noticed, it seems like it looses steps at the extruder motor. Even without filament the motor seems to go much slower sometimes when printing a line with constant extrusion. It seems not to be related to volumetric extrusion.
I suspect a timing issue - maybe some steps are to fast so the driver will not follow them? The effect is visible with the original motor as well as with a BondTech extruder.

gr5, 2019-01-30:
I have 17.10 and don't have any problems with underextrusion.  Please post your issue on the UM forum and add @gr5 to the post to catch my attention (or send me a direct message on um forum).  Maybe you could make a video showing how the extruder slows down?  Maybe something is wrong with the stepper driver chip?  We should continue this discussion on the forum.

Ricky1975, 2019-02-01:
Due to another defect I had to replace the mainboard.
Since the new board is in place, the effect is gone. Extrusion is normal and the stepper runs with the same speed all the time.
So I think, the issue might be related to timing with Generation 1 boards. As my old board is defect now, I have no chance to to reproduce the issue. Maybe somebody else has the same problem?

gr5, 2019-02-01:
Again, this should be on the forum.  Not here.  I have only gen 1 boards.  All my UM2 series printers (I have 4) are quite old.

I suspect it was a defective stepper driver.  I was going to suggest swapping cables on ext1 and ext2 or ext1 versus Z axis, etc.  But didn't want to talk about debugging hardware issues on tinkermarlin github.  I'd close this issue but other's have reported problems so I'll leave this open (although tinker may close it).