TinkerGnome/Ultimaker2Marlin #68: 17.02.1 Material change issue

jwitos, 2017-03-26:
Hi there! I was trying to change the filament and after choosing "change" option on the main menu, printer tried to home itself but it couldn't find the home on the X axis and tried to force move it over the limits. I couldn't reproduce this bug, however, since restarting the printer solved the issue. I'm using Ultimaker 2+ and this happened after several hours of printer being idle.

gr5, 2017-03-26:
It could be that something (like a fan on the side of the head) got in the way such that it had trouble hitting the limit switch.  Check exactly what is hitting the limit switch and how far you can push the head farther before it stops moving.  You might want to bend the limit switch such that it triggers 0.2mm sooner.  Or it could be that it has moved.

jwitos, 2017-03-29:
Unfortunately, I don't think that's the issue. I could home head with no problems, so the limit switch mechanically-wise should work properly. I had the same issue today, but when I was trying to print, I took a video of how the head hits limit switch: https://www.youtube.com/watch?v=XfM2qxeH9uk (noise warning ;)). And this time I had to factory reset to solve, but other than that - works fine