TinkerGnome/Ultimaker2Marlin #75: Print Area, Max Z cant go above 999 mm / EEPROM stuck

nbasse, 2017-08-01:
I'm trying to change the max Z position to 3000 mm from the default 3600 for the Z-unlimited.
In the Print area settings menu Z can't manually go above 999 mm. 
I have tried changing the value in configuration.h and reflashing but all the old values seams stuck in the EEPROM. 
I have tried M502, M501 and M500 with no luck.

Is there another way to force the EPPROM to reread all values from the firmware?

gr5, 2017-08-01:
reset to factory defaults

Something like that  - it's an option in the menus.  It will copy values like this from firmware to eeprom.  The basic problem is that the location where these values are stored in the eeprom change from one version to the next and usually the firmware handles the upgrade gracefully - copying values to the new location but not always.

so the safe thing is just "reset to factory settings" or whatever it's called.

gr5, 2017-08-01:
Acutally checking the print area settings in the menu *might* mess it up.  So instead try just moving to position Z=2000 maybe?

nbasse, 2017-08-02:
I cant find any Factory reset option in the menus. If you write exactly where it should be, maybe I can look again for the 10th time.

(I'm on the zunlimited branch if that changes anything)

nbasse, 2017-08-02:
Now i have also tried swithing to a custom compiled firmware from https://github.com/Ultimaker/Ultimaker2Marlin (had to remove heated bed sensor to get past error message) and from there choosing factory reset in the menu and then flashing the z-unlimted version again. 
When looking in print area menu it has changed a little bit; X was set to 0 mm as maximum, Y to 225mm  and Z is still 3600mm.

nbasse, 2017-08-02:
Now i found this section in UtiLCD2_menu_prefs.cpp starting on row 325.

After adding a 9, I now can change the value true the print area menu above 999 to 3000 as I wanted 😃 

`static void lcd_limit_zmax()
{
    lcd_tune_value(max_pos[Z_AXIS], 0.0f, +999**9**.0f, 0.1f);
}
`

I haven't learned git yet so I don't know how to suggest this change in the code. So someone who knows how to, please feel free to do so.

gr5, 2017-08-02:
Nice find :smile: 

@TinkerGnome - please note suggested code change in post just above.  Without this code change the Z limit can't go above 999mm (1 meter) and the z-unlimited should be able to go much higher than just 1 meter.

TinkerGnome, 2017-08-03:
27ff0310e1cab278e5117f6d19121542403ae217

closed 2017-10-19 by TinkerGnome