TinkerGnome/Ultimaker2Marlin #76: FW 16.08.2 - incorrect speed reading in the info menu during printing

Inschenjoer, 2017-10-05:
I used your FW for the first time today, TinkerGnome, and I can only guess how much work you must have put into this. A big thank you right from the start!

Now to my question: my UM2 with TinkerGnome FW (16.08.2) showed a printing speed of 28mm/s instead of the 50mm/s I entered in Cura (15.04.4) even though I had set the E-steps to 492.45 for my Bondtech extruder. Is there anything I can do within this FW to get a correct reading of the printing speed while printing? My UM2 seems to print correctly at the set speed, though, it only shows a wrong value during printing. (FW 16.08.2 was recommended to me by Bondtech as the newest stable version - and stability is crucial to me - so I would rather not upgrade to any instable or beta version.)

PS: Please write slowly when answering me, I'm not an expert... ;-)

TinkerGnome, 2017-10-07:
Well, it just displays the value that the motion planner has written into the current block as 'nominal speed'. Was this slow enough? 😃 
Is it possible, that Cura has slowed down the speed because of the 'minimum layer time'? If in doubt search for all lines with a G1 command and the parameter F (in the gcode file ). F is the feedrate in mm/minute, this way you can check if the displayed value is plausible in any way. All parameters are explained here: [G-code RepRapWiki](http://reprap.org/wiki/G-code#G0_.26_G1:_Move)

Inschenjoer, 2017-10-07:
Thanks a lot, TinkerGnome! And - yes, that was slow enough. :-)
I'll check the parameters. Any program you would suggest I could use to go through the gcode?

TinkerGnome, 2017-10-07:
I don't know a special tool. Any text editor with a search function should be sufficient for this task. This one can be used with regular expressions: https://notepad-plus-plus.org/

gr5, 2017-10-08:
Just regular notepad works fine but if you edit code every day then I also love notepad++

Inschenjoer, 2017-10-08:
So I checked my gcode with notepad++, and it all looks plausible. Especially your hint at the minimum layer time setting, TinkerGnome, seems to explain the slowdown. Thanks both of you for your comments and your helpful and friendly way in which you dealt with a beginner like me.

closed 2017-10-08 by Inschenjoer