TinkerGnome/Ultimaker2Marlin #77: "X/Y switch broken" inconsistent error on UM2+

pbalden, 2017-10-08:
Thanks for the recent update.
I rushed to try it but printing fails on my UM2+ (actually UM2 + extruder upgrade).
It says "X/Y switch broken" and stops just after homing the head.
With the regular firmware it homes OK and goes on to print just fine.
Maybe some wrong setting with X/Y limits in this version?

Turtletrumpet, 2017-10-08:
I run the V17.10 on an UM2+ Extended. Third print is running(around 2 - 3 hours per print). One small bug. I print with octoprint. When the print finished, I can't send any command to printer. I manually disconnect and connect within octoprint and it works. I don't powercycle the printer(like V16.08.2) and start the next print. Running without problems. I'll keep watching my printer.

Many, many thanks to TinkerGnome for the work!

Regards,

Dirk

TinkerGnome, 2017-10-14:
@pbalden - if you switch between the regular and the tinker firmware on a UM2+ the stored settings in EEPROM gets corrupted. If you want to try it again, double check the settings in "Advanced -> Preferences -> Print Area" after the firmware install.

closed 2017-10-19 by TinkerGnome

pbalden, 2017-10-19:
Indeed, just installed version 17.10.1 and did a full Factory reset which probably sorted that out... and now everything seems to work. Thanks.