TinkerGnome/Ultimaker2Marlin #78: Tinker V17.10 Error - stopped temp sensor

IlnazAkmalov, 2017-10-09:
Hello dear friends! I really like the firmware from TinkerGnome. I saw the discovery of Tinker V17.10 and was very happy, decided to reveal my printer. But I had a problem. After making a decision, I decided to make Factory Reset. After the reset I adjusted the height of the table, then at the filament loading did not react to the encoder button. Tinker V17.02.1 was all right. Please tell me what you can do?
https://www.youtube.com/watch?v=uFkJesRr1vk
https://www.youtube.com/watch?v=3bNHeg0UkH4

yyh1002, 2017-10-09:
I had the same problem. Had to shut down and restart.

IlnazAkmalov, 2017-10-10:
can I roll back to the old version Tinker V17.02.1. there were no visible problems.

gr5, 2017-10-10:
You can definitely roll back.  But do a factory reset.  Each new install is designed to update the eeprom settings to where the new version stores them - each new version knows where the older versions stored the settings.  But they don't do this in reverse.  So do the factory reset after the firmware "downgrade".

IlnazAkmalov, 2017-10-10:
I through the menu made a reset (Factory reset). After this, this is the problem. What do you propose to do?

TinkerGnome, 2017-10-19:
@IlnazAkmalov this issue should be solved with [V17.10.1](https://github.com/TinkerGnome/Ultimaker2Marlin/releases/tag/V17.10.1)

closed 2017-10-19 by TinkerGnome

IlnazAkmalov, 2017-11-23:
All perfectly! Thank you very much!