TinkerGnome/Ultimaker2Marlin #79: Target nozzle temperature always zero v17.10

yyh1002, 2017-10-09:
Hi,
I'm unable to start any print because target nozzle temperature is always 0, no matter what material settings I set. 
It won't start printing even if I manually raise the temperature to 200 degrees. 
The target bed temp is working properly though. 
Every other nozzle heatup command such as load filament, move filament etc are all working fine. 

Thanks.

IlnazAkmalov, 2017-10-10:
We have similar problems with you. This is clearly a glitch in the firmware. Appeared after the exposure.

TinkerGnome, 2017-10-13:
Thanks for reporting! Ahh, i see - this would be an explanation for the "insert material" issue too.
@yyh1002 i assume, you're using UltiGCode and the material profiles on the printer? Can you double check, if the material settings are still correct (just to be sure)?
I'll have a look at it.

yyh1002, 2017-10-13:
@TinkerGnome Yes I'm using UltiGcode. I double checked back then, the material settings has no problem. Thanks for looking into this!

TinkerGnome, 2017-10-19:
@yyh1002 - i'm not able to reproduce this with my printer.
Which printer variant do you own? Can you export the material settings to the sd-card and share the file?

yyh1002, 2018-01-29:
I'm using Ultimaker 2+. Sorry I don't have the material setting anymore because I switched to v17.09 mark2 firmware and experienced no such problem.