TinkerGnome/Ultimaker2Marlin #82: Tinker V17.10.1 Problem: heater catridge stops working after 3h printing

Turtletrumpet, 2017-11-14:
Hello,

I use Simplify3d V3.1.1. with Octorpint and an Ultimaker 2+. After flashing with cura 15.04.6 I do a factory reset. I use a 40w heater catridge and raise the value in the preference menu from 35w to 40w.
Printing starts normally. After around 3h the heat catridge of the printhead stops heating, so the temperature goes down. The temperature sensor keeps working and the printhead keeps moving. Does any other have a similar problem?
Thanks for reading.

Regards,

Dirk

gr5, 2017-11-14:
You should get a "heater error" message after 30 seconds of the temp being more than 10 degrees below goal temperature.  Do you get this?  If not then this is a HUGE clue.  It would imply that the code that controls the heater and checks for heater error is not running at all.  if you aren't getting "heater error" then please disconnect your heater and then manually set the temp to 100C and count how many seconds it takes to get the "heater error" message.  Just to make sure that you have this feature enabled (it's possible to disable this feature or set it to trigger after 60 seconds instead of 30 seconds).

Turtletrumpet, 2017-11-14:
I got no heater Error. At the moment I print the same file with V16.08.2. I see a untypical temperature deviation. Normally the temperature is +/- 1.5°.
[image]
I will flash 17.10.1 again and do a "PID auto setting" after the print is finished.

Turtletrumpet, 2017-11-14:
Oops, and I will disconnect the heater catrdige and try to get error message ...

Turtletrumpet, 2017-11-14:
Hello,
I disconnect the heater catridge and set the nozzle temperature to 100°C.
On V16.08.2: Heater Error: ER03
On V16.08.2 via USB/Octoprint: Heater Error: ER03
On V17.10.1: Heater Error, ER03
On V17.10.1 via USB/Octoprint: Heater Error ER03
I don't set the power budget of the heater back to 40W after factory reset. Next I make an AUTO PID. 
Tomorrow I will print again with V17.10.1.
Any other Ideas?

gr5, 2017-11-14:
Okay.  This is fascinating.  So v17.10.1 is capable of getting a heater error if you disconnect the heater.  But on a 3 hour print you did, the temp dropped by more than 10C for more than a minute, right?  

That tells me something bizarre happened such that the printer was not running the temperature code to adjust the heaters about 10X per second like it should.  Yet it hadn't completely crashed and was still stepping servos in a controlled manner.  WTF?  @TinkerGnome  have ideas?

gr5, 2017-11-14:
Well for now @Turtletrumpet I recommend you stick with tinkerMarlin version 16.01. That still seems to be the most recent stable version.  Seems to be.  I mean I have been using newer versions just fine but most people seem to have these bizarre problems on versions newer than 16.01.

Turtletrumpet, 2017-11-15:
Hi, I'm not sure, if the temperature dropped by more than 10°C. I only notice that temperature control is off.

gr5, 2017-11-15:
Oh!  So it sounds like you just think the PID is a bit off.  Well this is relatively normal.  Even on my UM3 with UM cores and with a single color/material print (only using one core) I've seen the temp be 3C low half way through a long print.

Reading about PID on wikipedia and thinking about it a lot might help you tune these a bit.  I'd increase P and I by maybe 20% and probably leave D alone.

I (integration) is the one that should eventually (after a minute or two) bring the temp to the right value so you could just play with that.  P is a more immediate effect.  either of these being too high will cause the temperature to oscillate - P too high makes it oscillate faster and I being too high is a slower oscillation.  D is supposed to fix overshoot and oscillation but it can make it worse.  D is like the brakes.  If the temp is moving too fast, D will slow it down.  It's like a limiter on temperature swings.  D can make it take a long time though to change to another temperature.  Too much D can ironically cause oscillation also.

Turtletrumpet, 2017-11-15:
Thanks for the information. After work i fine tune the PID and print three parts whith a printing time of around 1 hour per part. No problems happens. I hope i can print tomorrow a part with around 3 - 4 hour printing time and finish the print normally ...

Turtletrumpet, 2017-11-16:
The tip to raise P and I is great! I increase P and I by 12 percent after running auto PID. The temperature looks more stable! Thanks! The 4 hour print test with v17.10.1 is running ....

Turtletrumpet, 2017-11-20:
At the moment my printer is running well. I have no idea, why the printer stops heating as described in the first post. When i run into trouble, when I run long prints(around 20 hours), i will leave a message. Many thanks for help and the development of the tinker firmware ...

closed 2017-11-24 by Turtletrumpet